kartaview/android #3: Images are Mirrored

martijnv-telenav, 2016-07-28:
_From @MikeTho16 on July 28, 2016 3:21_

The images in my latest upload are mirrored (i.e. what is on the left in reality is on the right in the image and vise versa, for example: [http://openstreetview.org/details/10557/14](url). Thus, it appears as if I was driving on the left (wrong side in US), when in fact I was driving on the right and words on signs and on the pavement also appear backwards. 

_Copied from original issue: openstreetview/openstreetview.org#23_

martijnv-telenav, 2016-07-28:
_From @kalmanbencze on July 28, 2016 6:42_

Hello, could you tell us what device did you use? Thanks.

martijnv-telenav, 2016-07-28:
_From @MikeTho16 on July 28, 2016 15:18_

@kalmanbencze Samsung Galaxy S3 running Android 4.4.2. I should note that it doesn't always produce mirrored images, only sometimes.

iav, 2016-08-12:
See the same with Samsung Galaxy Note 4. 
Not sure but possible I make photos with rotated up-down from normal position - camera near right hand, big downscreen button near left hand.

AleZena, 2016-08-16:
Same here with a Samsung S3 NEO. With the same smartphone orientation sometimes images are mirrored sometimes are not

vincentdephily, 2016-08-19:
Same here with Sony Z3 compact - http://openstreetview.org/details/12242/5

buro9, 2016-08-21:
Reached this via HN, but I'm pretty sure the answer is EXIM orientation as some phones have their sensors physically orientated to fit design constraints of the cases and this requires that the software writes EXIM orientation to correct it.

Background: http://www.impulseadventure.com/photo/exif-orientation.html

Example of some Go code that handles image orientation: https://github.com/microcosm-cc/microcosm/blob/ccfad45e40ab93a266f719dcc95edda6cca78bb2/vendor/github.com/microcosm-cc/exifutil/exifutil.go

kalmanbencze, 2016-08-22:
We already read and rotate the frames based on their exif data, the bug comes from the filter that is used for a 180 degree rotation (horizontal flip). The issue will be fixed soon.

kalmanbencze, 2016-08-22:
ok, the bug is fixed, instead of horizontal flip, the frame was flipped both horizontally and vertically to result in a 180 degree rotation.

closed 2016-08-22 by kalmanbencze

iav, 2016-08-22:
Will it change already uploaded photos?

https://github.com/openstreetview/android/issues/3#issuecomment-241423493
This mail was scanned by BitDefender
For more information please visit http://www.bitdefender.com/frams/

MikeTho16, 2016-08-22:
@kalmanbencze Thanks for fixing this.  I will test it in the next couple of days. However, there never should be any "flipping" (e.g. swapping left pixels for right pixels) only rotation.

kalmanbencze, 2016-08-22:
@iav unfortunately already uploaded photos will not be affected by this, they will need to be fixed through the editor most probably.

@MikeTho16 the hflip+vflip is more efficient that doing two transpose operations.
Imagine a matrix, doing the flips doesn't need to resize the buffers for the pixels when having width>height because we know that the width and height won't change.

You can test the new build once it's up, starting from Friday 26th.

kalmanbencze, 2016-08-30:
There's a fix coming (more feature than fix) for mirroring the wrongly saved frames. I'll notify this thread once its available.