kartaview/android #10: Can't upload an interrupted sequence

Xevib, 2016-08-01:
I have a sequence that i can't upload. When I open the sequence there are no photos and if I try to upload the aplication crashes.
How I can attach more information to help the debug? There is any log?

alexandrui-telenav, 2016-08-01:
Thanks for the help @Xevib 

@kalmanbencze please provide some Android specific ways to help us debug.

kalmanbencze, 2016-08-02:
Hi, do you have a samsung SM-G900W8 (galaxy s5) or a GT-I9195 (s4 mini) ?

Xevib, 2016-08-02:
@kalmanbencze yes,this problem was detecetd on the Nexus S5X

kalmanbencze, 2016-08-02:
Can you check if you have any local files in /sdcard/Android/data/com.telenav.streetview/files/OSV/ ?
You should see some folders named SEQ_0, SEQ_1 etc., if you have only that specific track stored on the device, check if it has video files in its folder. If you have found it please check the size of the file (or the size of these mp4 files) Most likely you have a file that is over 100mb that fails to upload because it cannot fit in memory. Thanks

kalmanbencze, 2016-08-02:
I can see here on crashlitics that your device tries to upload a 134mb video file, which crashes the app.
You will need to extract that track, the device simply cannot upload it because it doesn't have enough memory to load the file for the request.

Xevib, 2016-08-02:
@kalmanbencze the OSV dir seems to be empty.
[image]

kalmanbencze, 2016-08-02:
Do you use external sdcard or only the phones internal storage?

And also if the folder is empty and you dont have an sdcard (which you probably don't, since you use the Nexus 5X - or is it Samsung S5?), check in the application if there are waiting local tracks to be uploaded.

Xevib, 2016-08-02:
@kalmanbencze No, the Nexus 5x don't have sd card slot. This problem happens on the Nexus. On the upload I have this:
[image]

kalmanbencze, 2016-08-02:
could you try removing the app from the recent apps screen, then restarting the app?

Xevib, 2016-08-02:
@kalmanbencze I did it , I closed the app and opened it againt and have the same problem. Restarting the phone happens the same

kalmanbencze, 2016-08-02:
thank you for the help, I will fix this issue, and push it with the next update

james2432, 2016-08-15:
[image]
Also have this issue, but it seems to delete all the files related to the sequence in the data folder when an upload is interrupted

kalmanbencze, 2016-08-22:
Rest assured, the app deletes local data only after receiving the green light from the servers that it has been successfully uploaded.
Most probably the track should be visible at all in the screen above.

bogdans-telenav, 2017-12-04:
I will close this since is very old and many improvements have been done here. If you still encounter this issue in the latest version (2.1.5) please reopen.

closed 2017-12-04 by bogdans-telenav