kartaview/android #11: Application stops taking photos

Xevib, 2016-08-01:
The application stops taking photos after a while when I put the max resolution(12MP) on a Nexus 5x.
The same image apears on the screen and it blinks like when takes a photo but the counter doesn't increase

alexandrui-telenav, 2016-08-01:
@Xevib if you managed to upload the track can you check on the website if there are pictures taken when the problem was reproducing? Just to figure out if it was a bug in the counter or it didn't took pictures at all. We'll try to reproduce this nonetheless.

Many thanks

Xevib, 2016-08-01:
I will try to reproduce the problem as soon as posible and check if the problem it's the counter

iav, 2016-08-12:
I had same problem: start capturing, then switch to waze navigator and drive. On a traffic red light switch to capturer and see it stay with old picture. Ipress red button 2 times, see actual picture, and drive again...
Then I upload all to the site and see 2 short segments.

What should I do in future to provide you information that help you fix this bug?

kalmanbencze, 2016-08-31:
@iav @Xevib does the problem still occur? check the numeber of taken images, put the app in background, drive, open the app, check the number again. Is it increasing while app is recording in background? App being in foreground or background should make no difference whatsoever because the whole functionality is on a background service. Please check with the latest build.

iav, 2016-09-01:
Last try with 1.9.7.  Still see that - my last track captured only 22 shots. I look to screen on start. Then I swich to waze navigator - and no more shots.
Can I send you kind of a log file?

Xevib, 2016-09-01:
With the last update it works

kalmanbencze, 2016-09-01:
@iav yes, please send me the log text files from /sdcard/Android/data/com.telenav.streetview/files/ every log should have a timestamp in their name which could help you find the right one. Thanks

reinvented, 2016-09-04:
I'm experiencing the same thing, with version 1.9.7 on my Nextbit Robin (Android 6.0.1), but in my case I leave the app running and do not switch to other apps mid-track-capture. 

This happened, for example, with track www.openstreetview.org/details/17385/1 -- only 180 images were captured, but I drove much further than this, and without any interaction with the device the app stopped capturing tracks at this point. I was not using an OBDII device at the time. No errors were shown on the screen, and the app did not lock up or freeze, it simply stopped capturing.

kalmanbencze, 2016-09-04:
Could you send me the logs from that time frame? They are at sdcard/Android/data/com.telenav.openstreetview /files

reinvented, 2016-09-04:
I've [uploaded all my logfiles from yesterday's tracks](https://c.ruk.ca/owncloud/index.php/s/G0QyjxmD2HMMxrf). The http://www.openstreetview.org/details/17385/1 was the final one I recorded for the day.

iav, 2016-09-10:
https://www.4shared.com/folder/W9asgt2p/telenav1.html
There logs for today's driving. Sequence showed me on map inside app about 2/3 as it should be.
I can't upload it to server - app tell me "error creating sequence" when I press upload, but it seems to be unrelated error.

linuxes, 2016-09-18:
I'v got the same issue from yesterday, i had ~30km drive but just 2km are recorded and even positions of each taken picture is not on right place on map ( http://openstreetview.org/details/21758/1 ), on way back i tried to record again but counter doesn't increments (size of taken photos was either on 0) but app was acting like it's taking a photos

kalmanbencze, 2016-09-19:
@linuxes Hey Adam, could you send me the log files located at "/sdcard/Android/data/com.telenav.streetview/files/" ? Although we know about this issue, it would be helpful in fixing it.