kartaview/android #8: Android app isn't on F-Droid

kalmanbencze, 2016-08-01:
_From @redsteakraw on July 28, 2016 2:9_

[F-Droid](https://f-droid.org) is the free software app store for android, it is a collection of various open source apps including many OpenStreetMap apps. I would love to be able to download and use the app from F-Droid.  

https://f-droid.org

_Copied from original issue: openstreetview/openstreetview.org#22_

kalmanbencze, 2016-08-01:
We will look into it, most probably will release on F-droid as well.

kalmanbencze, 2016-08-01:
_From @james2432 on July 31, 2016 7:6_

Should probably be moved to android repo @martijnv-telenav

pizzamaker, 2016-09-22:
Just want to add that F-Droid inclusion is important security-wise, as they build the app locally with their signing key.

More details on how to add it to F-Droid: https://f-droid.org/contribute/

kalmanbencze, 2016-09-23:
@pizzamaker the only closed source dependency that remains is the map sdk that we're using. The Google services are now optional so that can be excluded from the build. We're looking into alternatives for the map sdk currently.

rugk, 2016-09-23:
You seriously used Google services???! And a closed-source map sdk for contributing to such an open database like openstreetmap??

Oh, how good that there is FDroid. 

As for other SDKs I though you know your [own wiki pages](https://wiki.openstreetmap.org/wiki/Frameworks) and personally I'd recommend [MapBox](https://www.mapbox.com/).

kalmanbencze, 2016-09-23:
@rugk it's a bit  more complicated than that. Our initial choices were limited.

pizzamaker, 2016-10-14:
@kalmanbencze What map sdk are you currently using? Is there no other equivalent sdk in terms of quality?

rugk, 2016-11-10:
Any news here?

kalmanbencze, 2016-11-10:
Still ongoing, there are no good alternatives as of now that are approved. Also our focus was on the next big feature release.

kalmanbencze, 2016-11-10:
The google services are now optional. the only issue is the map sdk

gdt, 2017-01-30:
Probably a new issue should be created for each non-free dependency.    I haven't run the app yet, but mapsforge is used by SatStat and seems to work decently, plus it works offline.

derSoe, 2017-03-23:
Is there any progress regarding this issue? I'm very interested because I'm using OSC on a phone without Google stuff. That means I'm sticking on version 1.9.9.6 because this is the one of the apk you put into dropbox. The latest version actually must be 2.something. So this is a bit unsatisfying and I hope you'll find a solution soon.

kalmanbencze, 2017-03-23:
Hello @derSoe , I'm sorry about the inconvenience, I've updated the build on dropbox.
https://www.dropbox.com/s/fen2ibvyte10xhs/app-release-2.0.2.apk?dl=0
About fdroid I'cant promise anything because we have so many other priorities right now, and for the app to be accespted, we would have to find alternatives for a lot of components.

derSoe, 2017-03-23:
Thank you for the fast response. Of course Fdroid compatibility is not a major issue for you. With an up to date version on Dropbox people like me can life for quite a while I think. If you announce new releases on Twitter or somewhere else we can update the app manually.

Maybe you should also update the link on the homepage to avoid further confusion ;)

By the the way, that whole thing you're doing here is pretty awesome! Keep up the good work!

kalmanbencze, 2017-03-23:
Updated the website, it should now download app-release-latest which is updated everytime after releases. Until now it was pointing, in a static static way, to 1.9.9.9 I think.

rugk, 2017-03-23:
No FDroid is a big concern and I certainly won't install any app from DropBox.
If you have proprietary components in your app this means the app is not completely open source, so you should fix this anyway. It has a good reason that this is a requirement for FDroid applications.

kalmanbencze, 2017-03-23:
So because we're using Dropbox it's a spyware? Oh boy I wish there would be market where you could download verified apps safely... 
But seriously though, for the time being this would be the only alternative for people who don't use play services, and rest assured it's the same digitally signed apk. 
The only things keeping us from uploading to FDroid are the map sdk, the play services which we are using for some features of the app like Google login, the facebook api needed for Facebook login, and the crash reporting sdk from fabric. Removing all these from the source code leaves us with an app which has much less features and an app that we cannot provide support. Not to mention another parallel branch to maintain. 
While there are more pressing issues, we have to prioritize our efforts, but putting the app on fdroid is something we eventually want to do.

n76, 2017-03-23:
There are a couple of open source projects I know of what have created their own signed F-Droid repositories and distribute their APKs that way. On their websites they have a page with QR code and text giving the repository information (location, public signing key info, etc.).

It is easy to add those repositories into the F-Droid app and then you can easily install the project's signed APK on your phone and you are notified of updates, etc.

Basically, by adding the developer's repository to the F-Droid app you are telling the app (and phone) to trust the developer. If the developer is using libraries the F-Droid project doesn't like, no big deal as the F-Droid project is not building or signing the APKs. It also means that the developer controls what version is released and when it is released instead of the F-Droid project noticing a new release and getting around to maybe building it sometime.

rugk, 2017-03-23:
> So because we're using Dropbox it's a spyware?

No that does not matter. Would be the same for OneDrive, Google Drive or own website or so… It's just not really FLOSS, that's the point.

And I know a lot of projects who distribute builds without play services and not-using Facebook or Google login is not really a big deal and likely understood (or not used at all) by FDroid users. As for the fabric sdk, yeah, it's not a disadvantage for the user, but if you want to use such a thing switching to a FLOSS alternative might be a good idea.

kalmanbencze, 2017-03-23:
Yeah I know it's not the best way to distribute software, I was half joking, but until we have everything set up for an FDroid release its better than nothing I think.

davidak, 2017-03-23:
@kalmanbencze it's better than not beeing able to use the software (or have to install Play store). Thanks!

picsi, 2017-03-29:
FYI, this is where the discussion is taking place: https://gitlab.com/fdroid/rfp/issues/80

Currently the blockers are the JAR files at https://github.com/openstreetcam/android/tree/master/app/libs

smithfred, 2017-04-15:
While it might be tempting to say "F-Droid isn't important" (and that may be true in terms of potential users for many other open-source/libre apps), in this case, of an app designed to allow contribution to a free/open street imagery database, the sort of users who are interested in contributing are *much* more likely to be using F-Droid (and avoiding Play etc.) than average, since there will be a big crossover in terms of principles.

I would install and use this app *today* if it were free of closed-source dependencies (and available as a consequence on F-Droid). Looking at the imagery in my area (one of the biggest cities in the world), even my contribution would probably increase the available imagery in my city by 10%.

If OSC were a project with the scale of contribution that OSM has, a handful of potential extra users wouldn't be a big deal. But from what I see on openstreetcam.org, it isn't. F-Droid users are much more likely to be contributors since they already believe in a similar set of principles (a shared/open commons) enough to be using it instead of Play.

/0.02

p.s., why was the bug title changed to FDroid? The project's called F-Droid.

rugk, 2017-04-15:
> why was the bug title changed to FDroid? The project's called F-Droid.

BTW, I hate this syntax/spelling. Always get it wrong… (bad when you search for issues)

kalmanbencze, 2017-04-18:
I think you're absolutely right about F-Droid, I will try to elevate this issue, convince the people who need convincing.

IzzySoft, 2017-05-19:
I have to agree with @rugk on many things here (oh, hi rugk, nice to see you here :smile_cat: ). While the selected Map SDK might be a thing harder to fix (even though there's a good pointer to a replacement above), one thing very easy to replace is the Dropbox. Why not simply attach the `.apk` to its corresponding release (on the `releases/` page)? Not difficult to do, and much easier to find. If you need some help with that, please check the Github help on [Creating Releases](https://help.github.com/articles/creating-releases/), where this is described in more detail.

Looking forward to find the `.apk` at the new location then :smiley:

kalmanbencze, 2017-05-20:
Hi @IzzySoft , the map sdk is being used mostly because of company "politics" mostly, if it would be my call I would swap it out. Attaching the apk to the release would be ok, but since we're working on another repo, I need to set up an automated system for commiting here, but its complicated since not everything is open source, that is why there are so few releases marked on the project. Creating releases with the apk but not updating the code would be inconsistent. So yeah, it's complicated, it's ugly, it's what we have right now. I'll try (even harder than !!!1hard!1!!!) to make the release system before adding binaries inconsistently to releases.

IzzySoft, 2017-05-20:
@kalmanbencze Conclusion: Having non-FOSS components in your app causes tons of trouble :shit:

Just let me know whenever you managed to attach an `.apk` here to the `releases/` page (and think you will do so in the future). I could then setup my grabber so your app were available via [my F-Droid compatible repo](https://android.izzysoft.de/repo) (where "compatible" means it works with the F-Droid client, not that it's compliant with the F-Droid FOSS definition: my compliance criteria are a bit laxer, considering my repo a complement to the official one and a compromise between missing out too much and having to revert to Playstore too often :stuck_out_tongue_winking_eye:)

rugk, 2017-05-20:
Could we finally also come back to discuss whether/how this app can be included in the official F-Droid repo?

IzzySoft, 2017-05-20:
@rugk that would be the best thing, sure. Though reading the last post of @kalmanbencze this is rather unlikely:

> not everything is open source, that is why there are so few releases marked on the project

points to more non-FOSS components which cannot be (easily) open-sourced.

> we're working on another repo […] not updating the code [here at Github]

Afraid there's not much @kalmanbencze can do unless the company gets committed. In that context I share your mystification on why the app contains so much proprietary stuff if the goal is contributing to a FOSS project (no rant, just citing a fact).

rugk, 2017-05-20:
Indeed, very strange project and actually not better than [some contributors](https://github.com/openstreetcam/openstreetview.org/issues/60).
I think this issue can be closed. Unfortunately. This way, this project will never succeed.

kalmanbencze, 2017-05-21:
The closed source dependencies are the ones listed before, there aren't a lot of releases pushed here  because it's hard to maintain 2 versions of an app, one with map related functionalities and one without. It's just that we have code mixed in with calls to closed source map sdk. I could make the releases,  but only if I invest in reapplying the same teqnique, will the project be buildable. And doing this every sprint manually is just too much. 
Since the map sdk is legacy  we've done our part of patches on it, and we don't work with artifacts anymore, but code, and that cannot be easily

kalmanbencze, 2017-05-21:
Sorry about closing the issue, I'm writing on mobile,it was by mistake. 
I won't close it yet because slowly I'm going to get it done.

ramack, 2017-09-05:
Any news regarding f-droid? I'd like to check it out without to need to worry about updates...

bogdans-telenav, 2018-05-07:
Short update on f-droid compatibility.
I am not very familiar with the reason why the app is not f-droid compatible but our team has changed a lot in the last 2 moths so no work was made on this. Our new Android team will not have time to work on this since multiple users are experiencing problems with major components. The focus will be to increase the stability and usability of the app. There is an idea (regarding the map SDK) on how to make it work on F-Droid but we don't have the man power to do it.

ramack, 2018-05-07:
@bogdans-telenav thanks for your statement.
I fully can understand our standpoint. With limited resources (and I have NEVER seen a project where this was not one of the problems) you always have to prioritize. But in a collaborative project - and I'd say openstreetcam is one - it is also a question about the culture of the project and how you work together with "your" community and with the needs of your contributors. I personally will not contribute as long as I cannot install (and get updates "automatically") your app on a google-free phone. I think in projects in such a big scope it is already more important to care about the community and form its culture than to add new features.

Just my 2cents, I can perfectly live without OSC and I honestly wish you the very best!

derSoe, 2018-05-07:
@bogdans-telenav 

The point @ramack is addressing is a good one. Which users are you targeting? I think it's the nerdy, FLOSS loving type. Who else would work for free to contribute to a project such as openstreetcam? All the others simply would go to google streetview and be fine. But the nerdy mappers are more likely than others to have no google stuff on their phone. And exactly these people (including me) you are locking out. You should rethink your priorities.

n76, 2018-05-07:
This is still dragging on?

I posted [a workaround/solution over a year ago on this thread](https://github.com/openstreetcam/android/issues/8#issuecomment-288749683): Simply have Telenav setup their own F-Droid compatible repository with a QR code so adding it to a user's F-Droid app is easy. I've seen it done for developer releases on [Transportr](https://grobox.de/fdroid/), [Briar](https://www.briarproject.org/fdroid.html), [microG](https://microg.org/download.html) and the F-Droid app even comes with the [Guardian Project](https://guardianproject.info/fdroid/) repository available.

But I guess I should just go away as I've given up on this project: None of the phones I tried it on were apparently highend enough to keep up so even plugged into my vehicle's power they'd run the phones batteries down and were incredibly laggy. And all of my trial uploads ended up having the tracks (and thus photos) moved to adjacent streets making them worse than useless. My current "go to" is to just use my commercial off the shelf GPS equipped dash cam with some scripts to extract the GPS track and sequential images suitable for loading into JOSM. Unfortunately that makes my "personal street view" hard to share to others who might find them useful for editing features I don't care enough to map myself.

ec1oud, 2018-08-09:
I would like to see this get done too.  (I know it's some trouble; I also have an app that I'd like to get published there, and didn't quite figure out or invest enough time into that process yet.)

ramack, 2018-08-09:
If the APP doesn't include nonfree libs it ia not that complicated. If it helps I can support with that, but with Openstreetcam I think the effort is more on the side to get it fully free (as in software not as in beer). Or is there some progress in the meantime?

IzzySoft, 2018-08-09:
> as in software not as in beer

Uh, no. That would mean Google stuff is "free as in software" (iow: full of trackers). Nah, we don't want that. We want it "free as in free speech" (though some speeches make that strange as well :rofl: )

mvglasow, 2018-09-18:
@gappleto The APK is already available for download from Github—you can get the app even without Google Play, but you would need to check for updates manually.

@bogdans-telenav and others: could you post a detailed list of all proprietary libraries you are currently using, with the exact package name? That would make things a lot easier for first-time contributors who are not familiar with the code (as in: potential users who would happily use the app if we could get rid of the proprietary code inside, and who are willing to help make that happen).

If Telenav is not willing to drop certain functionalities offered by proprietary libraries and does not want to maintain two flavors of the app, maybe someone from the community is willing to maintain the FOSS branch. All we’d need is someone to take on the task, and your endorsement for a friendly fork.

ithink314, 2019-01-16:
@mvglasow Speaking of the APKs on Github, what is the meaning of "// Source code is not yet up to date." for the last 2 releases, 2.3 on Aug 8, 2018, and 2.4 on Aug 23, 2018? It seems the last available source code is for 2.1.6 on Dec 22, 2017. Is this going in the wrong direction, or am I confused?

mvglasow, 2019-01-16:
@ithink314 I guess I’m not the person to ask as I’m not a project member, just another individual who would love to see this app stripped of proprietary dependencies and available on F-Droid.

IzzySoft, 2019-01-16:
Currently, the last two versions of this app cannot be considered "open source". As @ithink314 pointed out, the last 3 tags all point to the very same commit – so there is no code openly available for the changes in the last two. Which makes me join the confusion voiced (if the code is not up-to-date, what are the APKs based upon?).

Availability on F-Droid (as asked for by @mvglasow) would be really nice, but fully depends on the source code being available (and the proprietary elements removed). And before you ask: I cannot take this into my repo; first because of the missing code, and second because it exceeds the size reserved per app there (20M).

rugk, 2019-01-16:
BTW, quite funny that GitHub also notices this and marks the first of these three releases with same commit as "Latest release" 😉:
[image]

https://github.com/openstreetcam/android/releases

That said, this app going closed source is anything but funny.

ramack, 2019-01-16:
I vote for honesty. Meaning to rename this baby to ClosedStreetCam.

Anyhow who would be willing to contribute data, code, ideas or anything else at all to such a thing? Those who don't care about freedom will happily use google streetview...

davidak, 2019-01-16:
There is no point to have ClosedStreetCam when there is Mapillary.

Please make this open source again!

mvexel, 2019-01-16:
Hi all,

We are aware that we haven't pushed code out and we'll work to mitigate this. Thanks for your patience.

james2432, 2019-01-16:
@davidak that's not true, even though the map display framework they use in the android app is not available, OSC allows you to retain copyright/control over your photos.

Mapillary owns the photo once it's submitted to mapillary, they can do what ever they like with it.

Read the two licenses and ghere is a difference in terms of legal rights.

OSC also has dedicated mappers that improves OSM data, where as mapillary gets the community to do it via "mapping events"

One example: OSC did turn restrictions in Canada and the USA(no one likes mapping those, but are very useful for routing. Each service has their advantage and disadvantage

IzzySoft, 2019-02-06:
Looking at the [`build.gradle`](https://github.com/openstreetcam/android/blob/master/app/build.gradle#L174), two-and-a half year after this issue was opened I still see no chance this can be included at F-Droid:

``` gradle
    implementation "com.google.android.gms:play-services-location:${ext.playServicesVersion}"
    implementation "com.google.android.gms:play-services-auth:${ext.playServicesVersion}"
    implementation "com.google.firebase:firebase-auth:${ext.firebaseVersion}"
    implementation("com.google.api-client:google-api-client-android:${ext.googleApiVersion}") { exclude group: 'org.apache.httpcomponents' }
    implementation "com.facebook.android:facebook-android-sdk:${ext.facebookApiVersion}"
```

WTF does this have to do with Facebook? Do you really expect people who value their privacy (and hence favor open source over proprietary stuff, and especially OSM over GMaps) will be happy with their data being transmitted to that Kraken (just as a reminder: that SDK transmits data before you can even ask the users agreement – namely right at the start of your app). In 2.5 years, you should at least have had the chance to remove that. Or is the privacy of your users irrelevant to you?

Then, a few lines later:

``` gradle
    implementation "com.crashlytics.sdk.android:crashlytics:${ext.crashliticsVersion}"
    implementation "com.crashlytics.sdk.android:crashlytics-ndk:${ext.crashliticsNdkVersion}"
```

 No, thanks. Finally:

``` gradle
apply plugin: 'com.google.gms.google-services'
```

Guess we'll have to close the [RFP](https://gitlab.com/fdroid/rfp/issues/80), as in that state F-Droid cannot build this app. And seeing the history of this issue, it doesn't look like these issues will be solved in the next 2.5 years.

Please let us know then, when finally you got rid of those trackers, so we can take another look.

Apologies if this sounds harsh; it's rather bitter disappointment.

Rudloff, 2019-02-06:
Note that the Facebook SDK might not be the biggest issue here (aside from privacy considerations) because [it is open source](https://github.com/facebook/facebook-android-sdk). (But it uses a custom license that is probably not DFSG-compliant.)

Crashlytics and Play Services are definitely a blocker, because they are proprietary libraries.

horatiuf, 2019-02-06:
Hi all, 
I am one of the Android developers who is working on OpenStreetCam.

The short answer to this request is that in the near future, 6 months or so, there will be no support for the app to be on F-droid.

The long answer is that in this period we are currently working to stabilize core features such as upload, player and camera by any means necessary, which includes refactoring, re-writing or bug fixes. This means that our resources are fully focusing on this and cannot look into this issues related to privacy which affect the app availability on F-Droid. When this is successful we will re-consider looking how to make the app available on this platform.

If you have any questions related to this feel free to post them.

IzzySoft, 2019-02-06:
@Rudloff I doubt it is [see here](https://gitlab.com/fdroid/fdroidserver/blob/master/fdroidserver/scanner.py#L78), our blacklist has `r'''["']com.facebook.android['":]'''`.

@horatiuf 

> cannot look into this issues related to privacy which affect the app availability on F-Droid.

Oh, those privacy issues affect your app in general, not only on F-Droid. Sorry to read you find privacy negligible. I guess with that statement you scare away more than one (potential) user.

tuxayo, 2019-08-19:
Now the bigger problem is that the app is not anymore libre/open source, see #153

IzzySoft, 2019-08-19:
Someone should notify the folks at OpenStreetMap Wiki to update [this page](https://wiki.openstreetmap.org/wiki/OpenStreetCam) then, as "open source Android app" no longer holds true (I bet the same must be said for the iOS app). And yes, this definitely disqualifies it for F-Droid. Sad, but dead.

mvexel, 2019-08-19:
The [wiki page](https://wiki.openstreetmap.org/wiki/OpenStreetCam) has since been updated to show that the Android app is proprietary. This doesn't seem quite correct either -- the fact that the source code is not easily available does not mean that the open source license does not apply. It should be, if you ask me, but it's not the letter of GPL.

For the record I am no longer affiliated with Telenav, just a concerned citizen, nor did I make the change to the wiki page.

IzzySoft, 2019-08-19:
> This doesn't seem quite correct either

Well, even that had been updated already, @mvexel – see the footnote there :smile: Amazing. And yes, I'm fully with you – but the source code not being easily available makes it impossible for F-Droid to build the app. And one shouldn't always stick to the letters too closely if that would mean losing the essence of it all. I see no reason to hide the code if the app is considered "open".

tuxayo, 2020-02-20:
> Crashlytics and Play Services are definitely a blocker, because they are proprietary libraries.

With the major [release of v3.0.1](https://github.com/openstreetcam/android/issues/153#issuecomment-589250886), how to check if Crashlytics and Play Services are still required?

mnalis, 2021-01-20:
look at [build.gradle](https://github.com/kartaview/android/blob/master/build.gradle#L9) and look for `com.google.gms`. Yes, unfortunately `4.0` [still requires non-free components to work ](https://gitlab.com/fdroid/rfp/-/issues/80#note_456673874)

[7 more replies]

closed 2023-10-17 by kalmanbencze