kartaview/android #13: No longer shows tracks around you

james2432, 2016-08-02:
Before I used to look at the map to see what was mapped and what wasnt in the android app to plan a rout to map unmapped roads, now the canvas is just blank
[image]

alexandrui-telenav, 2016-08-05:
hi @james2432. Can you please check if you're using the latest official version from Google Play?
Previous versions might have this problem but the latest should display the tracks.

james2432, 2016-08-05:
Does the beta version count? or should I just use the google play (public release)?

kalmanbencze, 2016-08-08:
Hi james2432,
Please install the official public release version from google play, since the beta was discontinued after the public release. Thanks

closed 2016-08-08 by kalmanbencze