kartaview/android #15: Start record is confusing

n76, 2016-08-06:
Open App, select the camera button and a red circle inside a white circle makes it appear that it is recording. But you have to tap that to get a rounded rectangle inside a white circle to actually record.

It would be clearer to me if the initial icon when going to the camera were a right pointing triangle. And maybe have it a different color than red.

iav, 2016-08-12:
Impossible to know is recording active or not just looking to screen.

james2432, 2016-08-15:
This is true, the smaller circle is hard to differentiate from the big circle, especially if you just want to glance up while driving to make sure its still recording

chrisdebian, 2016-12-26:
I agree, the only way I know recording is happening, is because the image/ memory count increases.