kartaview/android #17: Unable to connect to Bluetooth ODB2 dongle

n76, 2016-08-06:
Acquired the Bluetooth version of the recommended ODB2 dongle, a model VP11 "Mini ODB II Diagnostic Scanner" from VeePack. I am able to connect to it from my phone (Galaxy Nexus Maguro running a AOSP 6.0.1 ROM) and can access it from both ODB Car Doctor Free and Torque (Lite), so the dongle works and other apps can connect to it.

On OpenStreetView, after selecting "BLE" from "Connection type" and then "Connect" a blank white area appears with no list of ODB devices and no obvious select or cancel buttons (seems like they ought to show, perhaps greyed out, even if disabled for lack of items to select).

n76, 2016-08-07:
Some screen shots to illustrate the issue. If https://github.com/openstreetview/android/issues/14 is fixed then it is possible that I could figure out a patch/change to fix this and submit a pull request.

Images are:
1. Showing ODB2 device is paired
[image]

2. Showing that a different app can access data from the car via the ODB2 device
[image]

3. Showing that Bluetooth is selected in OSV
[image]

4. Showing the blank dialog for picking a device.
[image]

n76, 2016-08-07:
Additional, settings dialog of ODBII says "Select type of ODB module you use, Bluetooth, WiFi or BLE" but only options are for WiFi and BLE.

[image]

kalmanbencze, 2016-08-08:
Hi n76,
Unfortunately the dongle you have purchased is a bluetooth dongle, but in the app only Bluetooth low energy dongles are supported right now.
With BLE you dont have to pair in the settings, only connect to them when they show up in the popup.

We plan on adding normal Bluetooth dongle support, but currently only Wi-Fi and Bluetooth Low Energy dongles are supported. Hopefully well enable traditional Bluetooth dongles support also in the upcoming build.
Thanks, Kalman.

n76, 2016-08-08:
Perhaps the Wiki page at http://wiki.openstreetmap.org/wiki/OpenStreetView/OBD2 can be updated to indicate that only BLE and WiFi are supported at the moment. And maybe another column added to the table to show the connection technology of each tested unit.

Also, something ought to be noted about the Amazon link may lead to additional units, but only the one listed under "Model" has been tested.

I guess I was not careful, I simply clicked on the Amazon link on that Wiki page and then ordered one of the units that was on the Amazon page.

kalmanbencze, 2016-08-08:
Yes we should update it, but don't worry, your dongle will work fine once we push the update with the normal bluetooth support.

mvexel, 2016-08-08:
@n76 I updated the wiki. I'm sorry, this should have been much clearer. I'm happy to refund you the $ you spent on the dongle, email me to organize that.

n76, 2016-08-08:
No problem. I'll wait for the push with the update mentioned by @kalmanbencze 

Though it would be nice to see the source code pushed to GitHub. It might allow others to help on fixing/extending the app.

martijnv-telenav, 2016-08-10:
#14 we're working on getting that out there @n76 !

martijnv-telenav, 2016-08-10:
Will leave this open until the new release fixes this for you

iav, 2016-08-12:
Are somewhere obd2 BLE adapters? Could you please provide a link to sample?

alexandrui-telenav, 2016-08-12:
Yep. https://www.amazon.com/LELink-Bluetooth-Energy-OBD-II-Diagnostic/dp/B00QJRYMFC

kalmanbencze, 2016-08-12:
not a bug. optionally open different topic for improvement for Bluetooth OBD support

closed 2016-08-12 by kalmanbencze

iav, 2016-08-28:
#33