kartaview/android #18: Pictures left on phone after upload

MikeTho16, 2016-08-07:
After tracks have been uploaded data still remains on the phone at:
/storage/emulated/0/Android/data/com.telenav.streetview/files/OSV/

kalmanbencze, 2016-08-08:
Hi MikeTho16,
Could you be a bit more specific? What files remain? Does the app show you that there are interrupted uploads, or the osv app says no local recording but there are actually video files/track.gz files in the folder? Also do you use external sdcards?
Thanks.

MikeTho16, 2016-08-09:
@kalmanbencze I don't see anything in the app about interrupted uploads and I don't recall any upload being interrupted, although recording has often been interrupted by crashes. There are video files and track.gz files in the folders.  I am not using an external SD card. I will try and do more testing so I can provide you more specifics.

kalmanbencze, 2016-08-09:
Could you list the size of these files? separately for the subfolders, for the gz and mp4 files, I would really appreciate it.

MikeTho16, 2016-08-09:
@kalmanbencze Unfortunately I erased them as I was attempting to free up memory on my phone.  However, I will run some more tests. If it happens again I will provide the details.

kalmanbencze, 2016-08-31:
@MikeTho16 Does the issue still occur?

kalmanbencze, 2016-09-08:
closing the issue since no activity and nothing to follow up.

closed 2016-09-08 by kalmanbencze