kartaview/android #19: Images not in proper location

n76, 2016-08-08:
For example, see track file number 11585 and observe the location of the images being shown as on the freeway when the images themselves are clearly still on a surface street running roughly at a right angle to the freeway. The same, or at least a very similar, positioning error is evident when looking at the images using JOSM with the OSV plugin.

My phone doesn't have the best of GPS receivers, but I've never see it that far off when using any nav app nor when using things like KeypadMapper or OsmTracker.

This may be related to https://github.com/openstreetview/android/issues/16 which was about track file number 11498.

kalmanbencze, 2016-08-08:
Hi n76,
You are right, the track displayed on the site is incorrect, most probabli the position wasn't accurae enough and it got matched elsewhere on the map. We'll sort the problem out.
Thanks.

kalmanbencze, 2016-08-08:
This issue was moved to openstreetview/openstreetview.org#41

closed 2016-08-08 by kalmanbencze