kartaview/android #25: total upload size and processed no forgotten after minimizing

james2432, 2016-08-15:
I started an upload and switched to other apps as I had stuff to check, so I left it in background. When I come back the uploaded and total size are both 0.

[image]

alexandrui-telenav, 2016-08-16:
Definitely a bug, @james2432, added to backlog. Just to confirm, the upload continues and the % progress is updating, right?

james2432, 2016-08-16:
yes it continues to upload, just seems like the variables are not initialized on the restore procedure

kalmanbencze, 2016-09-01:
just fixed this issue, will be available in the next release (Septemeber 9th).

closed 2016-09-01 by kalmanbencze