kartaview/android #33: Support Bluetooth ODB2 dongles

afmenez, 2016-08-19:
It seems the app currently supports only Bluetooth LE ODB2 dongles.  I never seen one, I believe regular Bluetooth dongles are far more common (I have three different ones myself) and should be supported.

josemoliver, 2016-08-27:
I agree, Bluetooth (non BLE) dongles should be supported. Amazon has the [LELink Bluetooth Low Energy BLE OBD-II OBD2](https://www.amazon.com/LELink-Bluetooth-Energy-OBD-II-Diagnostic/dp/B00QJRYMFC) dongle, but I am not sure if it it supported.

kalmanbencze, 2016-10-17:
Done, we have support for All types of obd now, bluetooth BLE and WiFi, You just have to select the type of the adpater in the settings before connecting. It will be available in the next build.

closed 2016-10-17 by kalmanbencze

afmenez, 2016-10-17:
Nice!  I'll test it as soon as it is available.