kartaview/android #36: Some images uploaded upside down

reinvented, 2016-08-21:
I installed the Android app (v1.9.5) on my Nextbit Robin Android phone (Android 6.0.1).

I mounted the phone on my car dash, and took a drive through the city, [resulting in this track](openstreetview.org/details/13009/1). 

Things proceeded well until [the 9th photo in the track](http://openstreetview.org/details/13009/9), which was taken upside down, and had significantly poorer image quality.

The upside-down-ness continued until [the 52nd photo in the track](http://openstreetview.org/details/13009/52), when it reverted to right-side-up until [the 71st photo in the track](http://openstreetview.org/details/13009/71).

This flip-unflip-flip-unflip continued for the rest of the track.

Same behaviour noted in [the next track I recorded](http://openstreetview.org/details/13019/1).

As far as I can tell, nothing changed on the device, the camera mount, or otherwise that would have triggered this flipping, and the image on the phone screen, while the track was being recorded, was always right-side-up.

lorenzhs, 2016-08-22:
Duplicate of or related to #3 ?

kalmanbencze, 2016-08-22:
Most probably related, @reinvented  please test the next version (1.9.6 up)when it gets released, which has the fix for some orientation problems. Thanks

kalmanbencze, 2016-08-29:
fixed in version 1.9.6

closed 2016-08-29 by kalmanbencze