kartaview/android #38: String related

kalmanbencze, 2016-08-22:
put here all the bugs regarding strings, translation requests and more.

kalmanbencze, 2016-08-22:
The string "Connection type" that apears on settings>OBDII don't appear on the strings.xml

kalmanbencze, 2016-08-22:
FR DE IT ES RU translation

james2432, 2016-08-22:
French Translation of "Connection type":
"Type de connection"

kalmanbencze, 2016-08-22:
I will create a https://crowdin.com/ page for the translations, that way it is more ordered and can be exported instantly.

james2432, 2016-08-22:
https://www.transifex.com/ is free for opensource projects that are public in visibility

kalmanbencze, 2016-08-22:
Even better.

kalmanbencze, 2016-09-01:
Before I include the new translation, there are some more string that need to be translated, @james2432 @Xevib if you have the time, could you look at the strings.xml? thanks

james2432, 2016-09-01:
No problem, I'll create another pull request for missing strings

Xevib, 2016-09-01:
@kalmanbencze of course, no problem

stephan75, 2016-09-07:
You can also have a look at http://weblate.org that offers free hosting for open source projects, or have an own instance on own servers.

james2432, 2016-09-07:
If you check https://weblate.org/en/hosting/ their free hosting plan says this:
Source strings 	Negotiable
and it's a plan under their basic 1000, so at one point they could require you to pay for hosting. Transifex is free no matter how many strings you have and is used by many opensource programs like KDENLIVE (many strings translated)

kalmanbencze, 2016-09-07:
Hi I've already created a page on transifex, its free, I uploaded the latest data (with the fr and ca translations), it's ready to be translated.

https://www.transifex.com/telenav/openstreetview/

please those of you who would like to translate, register by clicking 'join team' at the link above.
Thanks.

kalmanbencze, 2016-11-04:
Translations are getting along nicely, we now have aproximately 6-7 languages (3 completely translated), guys if you have time you can check out and translate the new strings on transifex. Thanks.

james2432, 2016-11-21:
Finished French

kalmanbencze, 2016-11-21:
nice!

bogdans-telenav, 2017-12-04:
If someone feels like adding a new language please reopen

closed 2017-12-04 by bogdans-telenav