kartaview/android #40: Image upload progress not updated properly

grigoryk, 2016-08-23:
As I'm uploading my first data set, progress indicator updated only three times (at 20%, 42%, 71%), each time without any animations, and then slid nicely to 99% and finished.

Time remaining and upload speed both seems to be working correctly.

kalmanbencze, 2016-08-25:
This is because the upload progress is refreshed after each video file (which are 30-40 mbs depending on the device), we are working on implementing a more granular stream upload, which will have more appropriate progress.Although the animation should be played on every stage. Thank you for reporting :)

kalmanbencze, 2016-09-07:
the progress is now updated properly after every 50-100 KBs. version 1.9.8 will have it

closed 2016-09-07 by kalmanbencze