kartaview/android #42: Profile totals are not updates after uploading a track

grigoryk, 2016-08-23:
After uploading a first track, "My Profile" lists it correctly (with a proper distance), but totals display all 0. See screenshot.

[image]

kalmanbencze, 2016-08-25:
Hi @grigoryk, you are right, this happens because the track that you uploaded needs to be processed, which can take some time, depending on the size of it. We'll surely modify the statistics to reflect the current status accordingly.

kalmanbencze, 2016-08-25:
Did it appear in the statistics after a while? Do you see the correct statistics now?

grigoryk, 2016-08-30:
Yeah, just checked now (so about a week after uploading), and the stats appeared. Also, the reported distance of the track actually doubled to 4.3km (initially it was 2km).

bogdans-telenav, 2017-12-04:
I will close this since is very old and many improvements have been done here. If you still encounter this issue in the latest version (2.1.5) please reopen.

closed 2017-12-04 by bogdans-telenav