kartaview/android #44: Application use only GPS without Glonnas.

dmvy, 2016-08-24:
Phone: LG E960
OS: Android 6.0.1
When OSV running phone can't find more than 5 satellites. After exit StatusGPS find 12 or more satellites (GPS+Glonnas).

kalmanbencze, 2016-08-25:
We are using google services location in the app, and the app most certainly doesn't tweek the hardware GPS. Can you check if the same thing happens with other gps app (like Google Maps). But I don't see how this can be an issue, could you please clarify the problem?

kalmanbencze, 2016-08-25:
I don't think google services support glonnas, only the regular gps satellites.

n76, 2016-08-25:
Network location services (Google location services) should not effect satellite based location.

And, at least on the one phone I have access to that seems to use GLONASS in addition to GPS, there appears to be no way for the app or user to select one or the other: If you choose "device only" or "high accuracy" you get position from both GPS and GLONASS.

I've not used OSV enough to tell if the number of satellites seen is different in it than in other position using applications but it would surprise me if that were true: The Android API doesn't support that type of control.

closed 2016-08-31 by kalmanbencze