kartaview/android #46: Error creating sequence

oscarpg, 2016-08-25:
I have recorded a track (two sequences) but i cannot upload it. Every time I try it, I receive the error message "Error creating sequence" and the upload is cancelled. 

It seems that the upload does not starts, the percentage is 0% when I receive the error.
[image]
[image]

At the internal storage folder /Android/data/com.telenav.streetview/files/OSV I have two sequences, and inside them, the .mp4

[image]
[image]

My device is a Xiami Redmi 3, with MIUI 7.1 (Android 5.1.1)

Maybe this issue is related with #10

kalmanbencze, 2016-08-25:
Hi @oscarpg , do you also have a "track.gz" file in the recordings' folder next to the mp4 files? that is essential to create the sequence on the backend.

oscarpg, 2016-08-25:
Hi @kalmanbencze , Indeed, I have "track.txt.gz" for both sequences in their folders.

kalmanbencze, 2016-08-25:
could you check if you have any log .txt files at /sdcard/Android/data/com.telenav.streetview/files/ ? if there are text files named like "log_05_08_05_23.txt" please send me (kalman.bencze@telenav.com) those files so that I can check why the requests are failing... thanks

kalmanbencze, 2016-08-25:
Also, how large are the track.txt.gz files?

oscarpg, 2016-08-25:
I don't have any log file at that path... Can I activate the app logs?
The biggest trzk.txt.gz file is 818 KB. The txt file has more than 119.000 lines. It starts with:

> Xiaomi Redmi 3;5.1.1;1.1.1;1.9.5
> 1472103945.618;;;;;;;;;-1.0286865;-0.111083984;-0.347229;;;;;;;;
> 1472103945.619;;;;;;;;;-1.0286865;-0.111083984;-0.347229;;;;;;;;
> 1472103945.619;;;;;;;;;;;;;2.9886336;;;;;;
> 1472103945.619;;;;;;;;;;;;;2.9886336;;;;;;

I don't know if it is important, but when I've recorded the track, the mobile phone was offline, without internet connection.

kalmanbencze, 2016-08-25:
i will enable the logs in the next build later today. if you cant upload the tracks, you can find the logs at the location I specified.
About the network connection, it is not needed at all while recording, only when you are uploading, check if the connection is all right, check if you have internet before you try to upload.

oscarpg, 2016-08-25:
Perfect @kalmanbencze , let's wait for the next buid

kalmanbencze, 2016-08-29:
Hello @oscarpg you can check for the log files with the latest app version (1.9.6). Please do the following:
-remove the app from the recents screen
-start app
-try to upload
-if it fails, send me the latest log file from /sdcard/Android/data/com.telenav.streetview/files/
thanks

mkyral, 2016-09-02:
Hi,
I just created my first two test sequencies with OSV and have the same problem.
OSV 1.9.7 on Nexus 5, android 6

Log file is listed below. Maybe is the issue that my OSM user name is [Marián Kyral](http://www.openstreetmap.org/user/Mari%C3%A1n%20Kyral)  - with "á" and space. Hope it helps.

```
1472826277914 D/OSVApplication: Crashlytics: initialized
1472826277919 D/OSVApplication: onCreate: architecture is armv7l
1472826277986 D/UploadManager: setEnvironment: openstreetview.com/
1472826278058 D/SequenceDB: interruptUploading: number of reset UPLOADING statuses = 0
1472826278059 D/SequenceDB: interruptUploading: number of reset INDEXING statuses = 0
1472826278072 D/Sequence: refreshStats: sequence 0, count: 136
1472826278076 D/Sequence: Sequence: Sequence (id 0 images 136 from 136 status 0 number of videos 3)
1472826278078 D/Sequence: getLocalSequences: reverseGeocode: 
1472826278081 D/Sequence: refreshStats: sequence 1, count: 15
1472826278084 D/Sequence: Sequence: Sequence (id 1 images 15 from 15 status 0 number of videos 1)
1472826278085 D/Sequence: getLocalSequences: reverseGeocode: 
1472826278088 D/UploadManager: consistencyCheck: starting
1472826278095 D/OSVApplication: consistencyCheck: done.
1472826279417 D/MapFragment: onCreateView: 
1472826279422 D/MainActivity: onBackStackChanged: MapFragment 
1472826279426 D/LocationManager: startLocationUpdates: successfull: false
1472826279444 D/LocationManager: connect: connecting google api 
1472826279454 D/UploadManager: cancelListTasks: cancelled map list tasks
1472826279458 D/LocationManager: startLocationUpdates: successfull: false
1472826279459 D/LocationManager: connect: connecting google api 
1472826279467 D/LocationManager: connect: connecting google api 
1472826279469 D/CameraManager: open: Camera
1472826279574 D/CameraControlsFragment: Create mShutterButton
1472826279575 D/CameraPreviewFragment: addCameraSurfaceView: 
1472826279576 D/UploadHandlerService: addUploadProgressListener: MainActivity
1472826279597 V/CameraManager: Camera is open
1472826279607 D/CameraManager:  resolutionWidth: 3264 resolutionHeight : 2448
1472826279611 D/CameraManager:  Resolution of the image is: 3264 x 2448
1472826279615 D/CameraManager: getOptimalPreviewSize: using resolution: 1600 x 1200
1472826279616 D/CameraManager: enableHDR: false
1472826279621 D/CameraManager: updateDisplayOrientation: setting 90 to preview
1472826279623 D/CameraPreviewFragment: onCameraReady: 
1472826279626 D/CameraPreview: Using existing texture
1472826279631 D/CameraManager: updateDisplayOrientation: setting 90 to preview
1472826279632 D/CameraManager: safeStartPreview
1472826279705 D/MainActivity: version: {"version":0}
1472826279706 D/UploadManager: version: successful
1472826279768 D/LocationManager: startLocationUpdates: successfull: true
1472826280924 D/MapFragment: onSurfaceCreated: 
1472826282950 D/MapFragment: refreshDisplayedSequences: sending request
1472826283220 D/UploadManager: listTracks:  number of segments = 0
1472826287943 D/MainActivity: resizeHolders: ratio = 0.0
1472826287973 D/WaitingFragment: setupUploadButton: 
1472826287991 D/UploadHandlerService: addUploadProgressListener: WaitingFragment
1472826287998 D/WaitingFragment: setupUploadButton: 
1472826288001 D/MainActivity: onBackStackChanged: MapFragment WaitingFragment 
1472826288002 D/WaitingFragment: setupUploadButton: 
1472826288006 D/LocationManager: stopLocationUpdates: successfull: true
1472826293711 D/MainActivity: resizeHolders: ratio = 0.0
1472826293718 D/UploadHandlerService: addUploadProgressListener: UploadProgressFragment
1472826293734 D/UploadManager: initialize: mInitialUploadTaskSize = 4
1472826293740 D/UploadProgressFragment: onCreateView: newly creating views
1472826293744 D/MainActivity: refreshSignatureValue: signature 2/2
1472826293745 D/UploadHandlerService: addUploadProgressListener: UploadProgressFragment
1472826293746 D/MainActivity: onBackPressed: WaitingFragment
1472826293747 D/MainActivity: error: popBackStackImmediate failed
1472826293747 D/MainActivity: onBackStackChanged: MapFragment WaitingFragment UploadProgressFragment 
1472826293751 D/LocationManager: stopLocationUpdates: successfull: true
1472826293794 D/WaitingFragment: setupUploadButton: 
1472826293799 D/UploadProgressFragment: onUploadStarted: totalSize = 73215552
1472826293848 D/UploadManager: commitNextSequence: 2
 -------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------
1472826295439 D/UploadManager: createSequence: {"status":{"apiCode":"This value does not match the expected UTF-8 charset.","apiMessage":"(username)","httpCode":400,"httpMessage":"Bad Request"}}
1472826295441 D/UploadManager: {"status":{"apiCode":"This value does not match the expected UTF-8 charset.","apiMessage":"(username)","httpCode":400,"httpMessage":"Bad Request"}}
1472826295465 D/UploadManager: commitNextSequence: 1
 -------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------
[... cut ...]
1472826295734 D/MainActivity: onBackStackChanged: MapFragment WaitingFragment 
1472826295735 D/WaitingFragment: setupUploadButton: 
1472826295737 D/LocationManager: stopLocationUpdates: successfull: true
1472826326535 D/CameraManager: setOrientation: camera rotation set : info.orientation(90) + orientation(0) %360 = 90
1472826326538 D/CameraManager: updateDisplayOrientation: setting 90 to preview
1472826328486 D/CameraHandlerService: removeCameraSurfaceView: 
1472826329175 D/LocationManager: disconnect: disconnecting google api
```

oscarpg, 2016-09-02:
It seems to be exactly the same. My logs had just that error:
`1472461637005 D/UploadManager: createSequence: {"status":{"apiCode":"This value does not match the expected UTF-8 charset.","apiMessage":"(username)","httpCode":400,"httpMessage":"Bad Request"}}
1472461637010 D/UploadManager: {"status":{"apiCode":"This value does not match the expected UTF-8 charset.","apiMessage":"(username)","httpCode":400,"httpMessage":"Bad Request"}}
1472461637049 D/UploadManager: cancelUploadTasks: cancelled all upload tasks`

And my username is "**Ó**scar PG"

kalmanbencze, 2016-09-02:
You are right this seems to be the problem. Well fix this Monday ASAP

kalmanbencze, 2016-09-08:
ok, with version 1.9.8 and up it will be fixed

closed 2016-09-08 by kalmanbencze

iav, 2016-09-10:
Have this error just now. 
Where 1.9.8 will be released?
Will this sequence upload after I install 1.9.8, or I have to delete this sequence?

kalmanbencze, 2016-09-11:
It will upload, the update is on the way. Officially it would have been Friday, but had a drawback, so only tomorrow.

kalmanbencze, 2016-09-12:
Ok, the update is live, you can update the build and upload your tracks.

mkyral, 2016-09-12:
Thanks. I just installed the update. No sequence create error, but now it is uploading and uploading and still 0 MB uploaded. Don't know the reason, maybe a slow wifi. Will try it evening from home.

kalmanbencze, 2016-09-12:
Please do so, even if you have a slow connection, depending on the total data size, it should increase after a while. If you encounter any more issues, just notify me on this thread. Thank you

oscarpg, 2016-09-12:
@mkyral, I have the same problem. It seems that there is a nullPointer:
```
1473672510042 D/UploadManager: initialize: mTotalSize = 87665715
1473672510073 D/UploadProgressFragment: onCreateView: newly creating views
1473672510075 D/MainActivity: refreshSignatureValue: signature 1/1
1473672510079 D/MainActivity: onBackPressed: WaitingFragment
1473672510080 D/MainActivity: error: popBackStackImmediate failed
1473672510087 D/MainActivity: onBackStackChanged: MapFragment WaitingFragment UploadProgressFragment 
1473672510095 D/GoogleLocationManager: stopLocationUpdates: successfull: true
1473672510139 D/WaitingFragment: setupUploadButton: 
1473672510147 D/UploadProgressFragment: onUploadStarted: totalSize = 87665715
1473672510165 D/UploadManager: uploadCache: Attempt to invoke virtual method 'byte[] java.lang.String.getBytes(java.lang.String)' on a null object reference
```

No error is shown at the app.

iav, 2016-09-12:
+1.
No upload, no errors in gui.
Do you need logs?

kalmanbencze, 2016-09-13:
Yes please, reproduce it a few times. So the progress screen is visible, it
just doesn't upload?

kalmanbencze, 2016-09-13:
Hey @iav , @oscarpg , @mkyral, try to  log out and log in again. this should fix the issue and should let you upload, lated today I will push a fix for this later today.
If you're curious, this was an issue with the security token that you receive from the server being null from the old session. Logging in again should fix it.

mkyral, 2016-09-13:
@kalmanbencze: Yes, it works after relogin. Thanks a lot.

kalmanbencze, 2016-09-13:
Thank you for reporting it. This needs to be fixed fast.

oscarpg, 2016-09-13:
Indeed, it works, thanks

iav, 2016-09-13:
Works, thanks.

https://github.com/openstreetview/android/issues/46#issuecomment-246599768
This mail was scanned by BitDefender
For more information please visit http://www.bitdefender.com/frams/