kartaview/android #47: Question more than bug

n76, 2016-08-27:
A comment by @kalmanbencze on a [bug about images being in the wrong place](https://github.com/openstreetview/openstreetview.org/issues/41), a quick glance at the code, looking at the files built up after running the app and remembering that my phone was getting very laggy and hot when this occurred leads me to wonder:

It appears that the app takes a sequence of images and uses ffmpeg to create a more compressed video sequence. Is this correct?

Does this conversion of single still images to video frames use the graphics processor or the CPU?

If it is designed to use the CPU then my old and slow phone might not be up to the task.

If the app (i.e. the ffmpeg library) is supposed to be using the graphics processor on the phone and isn't, then it could be an issue with the ROM I am running in which case I might be able to fix it by installing a different ROM.

kalmanbencze, 2016-08-29:
yes, ffmpeg indeed should be using GPU for the encoding and manipulation of the frames. what device/rom/kernel are you running? Have you experienced problems?

n76, 2016-08-29:
Its a Galaxy Nexus (maguro) running a custom 6.0.1 ROM with a 3.0.101 kernel. While I had not personally noticed it before, there have been reports by other users of occasions where the GPU stopped responding and a number of graphics operations were then done by the CPU at to cost of lots of speed, battery usage and heat. Sounds to me like this GPU problem may have occurred while I was testing OpenStreetView. Unfortunately, I didn't have a laptop and cable to do a logcat at the time to see in more detail what was happening.

Thanks for the information. Is there a discussion forum for this project somewhere that would be better for handling questions like these?

kalmanbencze, 2016-08-31:
there are some reddit pages dedicated to osm/osv, maybe there would be more apropriate.
https://www.reddit.com/r/openstreetmap/
https://www.openstreetmap.org/user/mvexel/diary/39274

closed 2016-08-31 by kalmanbencze