kartaview/android #48: Crashing on track recording with v1.9.6

reinvented, 2016-08-27:
With my Nextbit Robin (Android 6.0.1), I was able able to record tracks with version 1.9.5 without any crashing problems.

After updating to version 1.9.6 however, the app consistently crashes shortly after I tap the record button, before taking any images. There is no error message; the app simply crashes.

This happens whether I am connected to an OBD2 device or not.

I can email a crash log taken immediately after the crash if required.

james2432, 2016-08-27:
I also have this but only with OBDII

james2432, 2016-08-28:
Been trying to reproduce, but now i have good gps signal, when it was crashing it was yellow and red

james2432, 2016-08-28:
http://pastebin.com/gav6LAFD log file after crash, I only get it when gps precision goes red

Xevib, 2016-08-28:
I have the same problem. I removed the application data and the problem persists

jekader, 2016-08-28:
Got a similar problem with v1.9.6. It was a fresh install, I recorded the first track of ~10 photos successfully. It started crashing afterwards and after I attempted to upload the sole successful track the app crashed and won't start any more. I'm using a Sony Xperia z3 compact.

kalmanbencze, 2016-08-29:
I have reproduced the issue, found the source of the problem, we'll do some testing today on the fix before pushing an update later.
@reinvented 
@james2432 
@Xevib 
@jekader 
@josemoliver
I would like to ask you to test out the next version and confirm if it is fixed on your device as well.

jekader, 2016-08-29:
@kalmanbencze glad to help test this and provide logs if anything goes wrong!

kalmanbencze, 2016-08-29:
Unfortunately the crash occurs in natice C code, so no logs are available for you to send, unless you could extract logcat logs live with adb...

jekader, 2016-08-30:
Installed v1.9.7 and didn't get any crashes so far. Thanks for the good work and fast fix! Can anyone else please confirm if the new version in the play market works stable?

kalmanbencze, 2016-08-30:
Glad to hear it, hopefully the others will confirm it also.

james2432, 2016-08-30:
Works for me, had yellow and red precision and kept capturing

closed 2016-08-31 by kalmanbencze

reinvented, 2016-08-31:
I can confirm that this is working on my device now.