kartaview/android #51: Some images still upside down in v1.9.7, and image quality degraded

reinvented, 2016-08-31:
If you look at the track I recorded today using v1.9.7 at:

http://openstreetview.org/details/16542/1

You will see that the image quality starts off clear and sharp, with images properly oriented.

However at image number 28:

http://openstreetview.org/details/16542/28

the image is rendered upside down.

The following image, number 29, goes back to rightside-up:

http://openstreetview.org/details/16542/29

but from that point on in the track the image quality is significantly degraded (despite there being no change in conditions).

kalmanbencze, 2016-09-01:
I have checked the images, I think the issue is with the focus, could you try locking the focus on your next trip?
by tapping the preview somewhere, you force a focus event on the region that is tapped
by long pressing on a region in the preview you focus to that region and it remains locked to that focus point.

We've had problems before with the continuos focus, the device was focusing on the dashboard or even the windshield in some cases.

reinvented, 2016-09-04:
I've recorded another trip, http://www.openstreetview.org/details/17355/1

The image orientation issue no longer seems to be a problem, but the focus issue remained, despite my following your guidance to try focus locking (long-pressing on the screen). The first part of the trip is fine, but then, around http://www.openstreetview.org/details/17355/31, things start to go out of focus.

Is there any way that I can tell whether focus lock is active or not?

kalmanbencze, 2016-09-12:
I honestly recommend not locking the focus, because in most of the cases the phone's continuous auto-focus works just as it was intended. if you lock the focus and the enviroment changes to be out of focus, it will not refocus untill you explicitly tap somewhere.

bogdans-telenav, 2017-12-04:
I will close this since is very old and many improvements have been done here. If you still encounter this issue in the latest version (2.1.5) please reopen.

closed 2017-12-04 by bogdans-telenav