kartaview/android #53: Tracks not captured with ODBII connected (version 1.9.7)

reinvented, 2016-09-04:
Starting with v1.9.7, if I connect the app to my ODBII device (Veepeak WiFi OBD2 Scanner) on my Nextbit Robin (Android 6.0.1) device, tracks are not captured.

Under "Settings," I am able to connect successfully to the ODBII device, but the speed indicator on the track-capture screen never moves above 0 km/h, and I assume this is why no photos are captured.

I was able to successfully capture tracks with the same device and the same ODBII with v1.9.5 without issue.

I am able to connect other apps (Torque Pro, for example) to the ODBII device and demonstrate that it's operating properly.

kalmanbencze, 2016-09-08:
Unfortunately the obd connection has its flaws, what I recommend is to reset the connection, disconnect then reconnect through the settings, or even reconnect to the obd wifi connection.
This might be able to fix the 0 km/h issue.
from version 1.9.5 to 1.9.7 there arent any changes to obd funtionality.
It would help if you could post the logs from around the event. Maybe next time you encounter the problem.

bogdans-telenav, 2017-12-04:
I will close this since is very old and many improvements have been done here. If you still encounter this issue in the latest version (2.1.5) please reopen.

closed 2017-12-04 by bogdans-telenav