kartaview/android #54: Error creating sequence

Xevib, 2016-09-11:
When I try to upload the images I get an "error creating sequence"
I don't know if it can efect but I'm trying to upload 7 tracks that sum 2.4 Gb

kalmanbencze, 2016-09-12:
We had to make a production server deploy, which apparenlty broke the old builds' requests. Could you provide the log files from that period to help us fix it for the old builds? The 1.9.8 and 1.9.9 version work as expected, you can upgrade already.

Xevib, 2016-09-12:
With the last update I don't get the error creating sequence but I can't upload the images.
I attach all the log files
[log_09_12_09_44.txt](https://github.com/openstreetview/android/files/468270/log_09_12_09_44.txt)
[log_09_12_09_31.txt](https://github.com/openstreetview/android/files/468268/log_09_12_09_31.txt)
[log_09_12_09_30.txt](https://github.com/openstreetview/android/files/468267/log_09_12_09_30.txt)
[log_09_12_08_41.txt](https://github.com/openstreetview/android/files/468266/log_09_12_08_41.txt)
[log_09_12_08_06.txt](https://github.com/openstreetview/android/files/468269/log_09_12_08_06.txt)

kalmanbencze, 2016-09-12:
Does it happen all the time? What happens exactly? You press upload and the app crashes or it just exits the upload?

MapMick, 2016-09-13:
Hi,
I have the same situation!
With 1.9.7 I did get the error creating sequence and now with 1.9.9 the upload just newer starts, the screen stays at 0% / 0MB with nothing happening and no error messages!

kalmanbencze, 2016-09-13:
hey, @MapMick @Xevib , please do a logout then login again, I've checked the crash. I will push a fix for it, but in the meantime a logout-login should fix the issue.

Xevib, 2016-09-13:
After logout and login it works!

kalmanbencze, 2016-09-13:
Good then. @Xevib @MapMick Thanks for the help!

closed 2016-09-13 by kalmanbencze

MapMick, 2016-09-13:
Yes, after logout / login it also started to upload for me!

Thank's!