kartaview/android #55: several problems

fik236, 2016-09-18:
version 1.9.9
android 4.4.2

1) Can not upload images - that only displays that it is uploading - but nothing more (when I want upload - I am connected with wifi)

2) during recording - when phone go to sleep (display go to dark), it stops recording.

3) application often fall down (how can I store some log - to sen it?)

kalmanbencze, 2016-11-04:
Hello @fik236 , could you tell me what device are you using? The logs are automatically sent for Java crashes, for native crashes for now we don't receive reports, what you could do is get us a logcat directly from the device, by hooking up the phone to a computer and using the android sdk platform tools grab the logs with "adb logcat -d > log.txt" then post it here. eventually we will add crash logging from the C/C++ code as well.
More info: https://sites.google.com/site/androidhowto/how-to-1/save-logcat-to-a-text-file

fik236, 2016-11-07:
hello, you asking for phone type? it is cubot x6 - http://www.devicespecifications.com/en/model/494e2b39

now i have version of OSV 1.9.10. I can now upload images, but application is still quite a lot unstable. It does only about 40 images and then stop (it looks like loosing GPS signal (position pointer stops moving) - but if I try another  application it can show GPS signal) or whole application crashes.

kalmanbencze, 2016-11-07:
Could you send me some logs? Also I would suggest you to try the recording with another device, preferably one more high-end, as some low end devices running custom androids (poorly made by the manufacturer) have all sorts of kernel issues and memory leaks.

fik236, 2016-11-09:
And can you send me detailed instructions how to save logs? It must be precise manual - I am not android programmer.

kalmanbencze, 2016-11-10:
from your phones internal storage you can copy the log files, the path to it is /sdcard/Android/data/com.telenav.streetview/files/ here you will find several txt files. Reproduce the issues and send me the last one (by date). kalman.bencze@telenav.com
Thanks

bogdans-telenav, 2017-12-04:
I will close this since it is a very old report. If you still get this issues please file a new issue.

closed 2017-12-04 by bogdans-telenav