kartaview/android #56: Not possible to disable shutter sound

Enkhiel, 2016-09-25:
I'd really like to help taking pictures for OSM but driving 100km getting 2 shutter sounds per second really bothers me.
As soon as the sound can be disabled, I will continue taking pictures

kalmanbencze, 2016-09-27:
Hello @Enkhiel what device, and android version do you use? thanks.

Enkhiel, 2016-09-27:
Hey there,using a Xiaomi Mi Max running 6.0.1.

kalmanbencze, 2016-09-27:
This is a known issue on MIUI android, unfortunately this hasn't been fixed yet. We're working on finding a solution.

kalmanbencze, 2016-09-27:
Could you try to enter the camera app settings, then disable camera shutter sounds, under capture settings.

kalmanbencze, 2016-09-27:
or try to put the phone on silent mode...

Enkhiel, 2016-09-27:
tried DND and Silent without success :(I assume because it's the shutter sound and no special app-specific one.

kalmanbencze, 2016-11-04:
Unfortunately there not much we can do with MIUI for now. the sound effect is out of our reach.

closed 2016-11-04 by kalmanbencze