kartaview/android #57: Cannot upload:  "Error creating sequence"

jfburke3, 2016-09-26:
Samsung Galaxy S7 Edge, Android 6.0.1, Openstreetview Android app version 1.8.0.

I am unable to upload new sequences from my phone.  When I click Upload All, it starts going through the motions, then displays the message "error creating sequence" and stops trying.

maxolasersquad, 2016-09-27:
Same issue here. I've got a lot of images to upload, but I just get "Error creating sequence" when I try to upload.

kalmanbencze, 2016-09-27:
Hello @jfburke3 @maxolasersquad , this issue is fixed in today's version, 2.0.0
Please update the app from the google play store. Thank you.

maxolasersquad, 2016-09-27:
I'm on the beta channel, which seems to be stuck at v1.8.

kalmanbencze, 2016-09-27:
@maxolasersquad Yes, the beta was discontinued, I'm sorry. You need to download the release version from the market, just search for openstreetview.

jfburke3, 2016-09-28:
I missed that notice, too. 

If we install the new release over the beta version, will it recognize our current, non-uploaded sequences, and let us upload them? Or do we need to make a backup copy first?

kalmanbencze, 2016-09-28:
it will install next to the beta app, your data will not be lost , you will have two OSV apps parallel, you can try to upload your existing data with the beta, try logout then login, if it doesn't work you will have to upload it with the upload tool from the desktop. From now on use the release build from the market.

kalmanbencze, 2016-09-28:
Your tracks won't get deleted or anything, its just that the release build will be installed parallel, you could upload the tracks using the old beta build - if it works. I will try to update the beta build temporarily also, but can't promise anything. You should start using the release build.

kalmanbencze, 2016-09-28:
I just checked and unfortunately Google blocked our update, it might take a bit longer to release the new version.

closed 2016-10-17 by kalmanbencze