kartaview/android #59: Cannot connect to ODBII with BlueTooth

n76, 2016-10-27:
Issue https://github.com/openstreetview/android/issues/50 was closed indicating this feature was added.

Running OSV 1.9.10 on my Android 6.1.0 phone, I cannot connect to a BT ODBII dongle even though:

- BT is selected in the OSV settings
- The ODBII dongle is shown as paired with my phone in the system settings
- A number of car performance and diagnostic apps can successfully connect and show things like vehicle speed and fault codes.

kalmanbencze, 2016-10-28:
What is happening after connecting in our app? does is say connecting... or does it say connected but no values are read during recording?

n76, 2016-10-28:
It says connecting for a while and then goes back to not connected.

The other apps I have that use the BT dongle present a pull down list of BT devices that are linked/paired to my phone and allows me to choose the one to connect to. It's been a while since I setup those apps, but my recollection is that I did not even need to be near the car (i.e. in range of the BT dongle) to select the paired device.

I don't see anything in the OSV settings other than the "connect" and I expected that to show a similar list of paired devices when pressed but all I get is the "connecting" for a while.

kalmanbencze, 2016-10-28:
There is a bug, I cache the Mac address of the device to auto connect, and in that case the list isn't showing up after pressing connect. I should add a forget device button or just show the list every time. Until this is fixed a clean install or a "clear data" should fix it. Just make sure you upload your local data before it. And sorry for the inconvenience.

kalmanbencze, 2016-10-28:
If the cached Mac address fails, it never reaches the connected state so that the user could press disconnect - (which clears the cached Mac address)

kalmanbencze, 2016-11-04:
This issue is fixed, the new version will have it.

closed 2016-11-04 by kalmanbencze