kartaview/android #60: Dependency to Google Play

PanierAvide, 2016-11-28:
When I start the application, it wants me to update Google Play services. The problem is that I don't use a Google Account in my smartphone. OpenStreetView don't want to dismiss the message, and the application is consequently unusable. Is it possible to remove this message or the dependency to Google Play ?

reinhart1010, 2016-12-03:
It seems that Google Play Services is used for location. I may test this app with Google Play Services disabled by Lucky Patcher.

n76, 2016-12-04:
Not sure if it uses Google Mobile Services for location, but if you aren't running Google at all check out the [microG projec](https://github.com/microg)t: OSV has no complaints running with it and there are location provider plug-ins for it that use on-phone databases if you are worried about a network provider keeping track of your location.

PanierAvide, 2016-12-04:
Thanks for the tip, I will test and let you know if this works

PanierAvide, 2016-12-04:
I can't make microG services install or update correctly, as I use the default ROM on my Android phone and it doesn't allow signature spoofing (which seems to be a requirement to make microG services work).

kalmanbencze, 2016-12-21:
this seems to be quite a popular demand - total google independence. I'll see what we can do about this. the two types of cases we tested were "Gapps" on device and without. I'll see to it that this is improved. We should just drop it, since the google provided location is only for ui, the recordings use raw gps sensor data.

LuccoJ, 2017-06-14:
As far as I know, it is not necessary to depend on any proprietary Google components to access network-based location, unless one is using the API that Google now "recommends" (but the original, native AOSP API still works).
Of course, if there are no network location providers installing on the system (such as UnifiedNLP/MicroG or GApps), doing that will simply result in obtaining no valid location before a GPS fix, but that is up to the user and doesn't change the licensing status of the app itself, as no proprietary dependencies would be included.

polarbearing, 2018-08-30:
Any progress on this? I was just trying OSC on the android and still stuble over this problem. 
Would it be feasible just to split the camera part off for people who want to review the frames anyway and upload by script later?

alexandrui-telenav, 2018-09-03:
No actual progress @polarbearing . We're now discussing what involves and how can we make it happen.

tuxayo, 2020-02-20:
Can someone check if the new major version (v3.0.1) still require the Google Services?

Rudloff, 2020-02-20:
I tried 3.0.1 and it seems to work without Play Services (at least the main functionalities).

NoNamePro0, 2020-11-04:
Whats with Version 3.2.1? And which features doesn't work with Play Services?

ctag, 2021-06-23:
Hi,

I'm not certain this is related, but trying to use KartaView 4.3.0 on my phone that has no Play service or Gapps results in a "No GPS" warning while recording. And afterwards the upload button returns "You have no local recordings."

tuxayo, 2021-06-23:
@ctag It doesn't look like it would be related but we never know!
Does the GPS work in other apps?

ctag, 2021-06-23:
Understood, Google Play is just what came to mind when I saw the error. 

> Does the GPS work in other apps?

It does. I also made sure to give KartaView location permissions as well.

hamishmb, 2022-10-18:
Is this still a problem, or does Kartaview now work without Play Services?