kartaview/android #63: App crashes/stops taking pictures on "low" memory phones

james2432, 2016-12-05:
Not every phone has 2GB of memory, I've tested this on a 1GB ram phone (has about 700mb free after OS)
After about 70ish photos or so, the application will freeze then eventually close.
[av_player_log[1].txt](https://github.com/openstreetview/android/files/631998/av_player_log.1.txt)
[av_recording_log[1].txt](https://github.com/openstreetview/android/files/632001/av_recording_log.1.txt)
[log_12_05_12_58[1].txt](https://github.com/openstreetview/android/files/632002/log_12_05_12_58.1.txt)
[log_12_05_12_15[1].txt](https://github.com/openstreetview/android/files/632005/log_12_05_12_15.1.txt)
[log_12_05_12_14[1].txt](https://github.com/openstreetview/android/files/632006/log_12_05_12_14.1.txt)

I can run mapillary app with no problem, I think the issue has to do with compressing photos into mp4 and not having enough memory to do so. If it is a memory issue, would it be possible to offer a low memory option that doesn't encode to .mp4 and just saves the jpgs?

kalmanbencze, 2016-12-21:
Right now we are finishing with implementing a "safe mode" which target low end devices.
This mode will basically will turn on when the app detects low memory, crashes during the encoding process, or when the encoding fails. Then in this mode, the app reverts to jpeg format which is more stable and less resource hungry. It will be released first on the beta channel (not the old beta app, the current release app now has open beta testing at https://play.google.com/apps/testing/com.telenav.streetview) in the upcoming few days, until then check out the gamification score system and leaderboards :).

james2432, 2016-12-21:
If that "safe mode" is in the new beta version, then I will try it out as I have the beta version installed

james2432, 2017-01-12:
Jpeg mode works fine on 1.9.35 with these phones

chrisdebian, 2017-01-12:
Works well on Nexus 5X, Android 7.2.1

Chris

chrisdebian, 2017-01-12:
Typo, I meant 7.1.1.

Chris.

kalmanbencze, 2017-01-12:
great news. thanks for testing

closed 2017-01-12 by kalmanbencze