kartaview/android #67: Track stops recording at 1031 images.

chrisdebian, 2016-12-26:
Beta 1.9.27, Android on Nexus 5X. Today, the current track stopped recording after 1031 images or 1Gb. I can't see any settings that I could change, to prevent this, and my device still had loads of space for images.

james2432, 2017-01-12:
Might be an issue with phone? I've recorded over 2k images in a sequence on BLU Selfie 2(crappy phones I have to run in jpeg mode, no gamification or odbII)

chrisdebian, 2017-02-17:
I'm currently recording in JPEG mode, and no longer experiencing this problem.

Chris.

james2432, 2017-02-17:
Same jpeg mode fixes the performance issue

kalmanbencze, 2017-03-23:
The nexus 5X isn't such a low end device, but unfortunately the agressive memory management (native, but also the java) and hardware acceleration + the kernel quality leaves much to be desired. I strongly suggest using the Jpeg encoding on the nexus 5X. Its funny how the original nexus 5 still beats most "new" phones, even though it is a 4 year old phone.

closed 2017-03-23 by kalmanbencze