kartaview/android #68: Track Uploads Frequently Fail in Background

chrisdebian, 2016-12-26:
Uploading tracks, may take many hours, if not at work, where broadband is very fast. When at home, the broadband is much slower, and during the upload time, I need to do other things on my Nexus 5X. In order to do this, I put the upload into the background, and then return when I have finished checking mail, etc.

When I return to OSC, it asks if I want to exit the app. I select cancel, not 'yes', and then I have to restart the upload sequence.

Related to this, I think it would be better if the upload is done in much smaller batches, and this would get imagery uploaded (and available) much more quickly, and would overcome the frustration of huge uploads being restarted. I assume that uploads are compressed, before sending.

Thanks for any future improvements.

Chris.

james2432, 2017-01-12:
I have this problem in the beta version, application upload fails and I have to resume the upload(you dont need to upload what was already successful, but it's irratating making a 4h upload into a 9h upload process with constant monitoring)

james2432, 2017-01-12:
Did a test with screens on and in foreground and the app eventually crashes and needs to be restarted. I will post logs once my uploads finish.

james2432, 2017-01-12:
[logs.zip](https://github.com/openstreetcam/android/files/701819/logs.zip)
These are the logs that were created during upload, I left a recording log out as it's 25mb in size and irrelavent

chrisdebian, 2017-01-22:
Still an issue in latest Beta.

james2432, 2017-02-03:
Still Present in v2.0RC

kalmanbencze, 2017-03-23:
Hello guys,
This issue was a nasty one, but I managed to fix it yesterday. There was an overflow of native handle object ids, shared between thread ids and DB cursor ids, because of a creating looping threads one after the other, but without cloing them. and android limits the native handles to 1024.
This week we are going to push the fixed build to the beta or the release - this remains to be seen.

BR,
Kalman

closed 2017-03-23 by kalmanbencze

james2432, 2017-03-23:
Would explain why it would always fail after ~500 pictures