kartaview/android #69: Error Generation and Reporting

chrisdebian, 2016-12-26:
Today, an error report briefly flashed up on my screen, when I used the 'back' button. This would be useful for the devs to look at, and perhaps functionality could be incorporated, which allowed users to submit the output, for investigation.

Cheers,

Chris.

chrisdebian, 2017-03-12:
Not sure whether any one looks at these bugs/ feature requests, but I think the application would reach maturity much more quickly, if automated bug reporting was enabled, or at least make it an option, when an error is generated.

Thank you,

Chris.

kalmanbencze, 2017-03-23:
Hi @chrisdebian , the crash logs and errors get logged through fabric crashlitics, together with relevant information, where we can se statistics and ultimately fix them. Besides this, we also planned a manual error reporting mechanism, which send us the relevant logs from the app.
So rest assured as long as you connect to the internet after an error or crash, we get the automated report.

Thanks,
Kalman, 
android dev

closed 2017-03-23 by kalmanbencze

chrisdebian, 2017-03-23:
Great!

Thank you.