kartaview/android #72: Should disable auto-focus by default

rdoeffinger, 2017-01-07:
I can't really imagine in which case auto-focus would be really useful, most part of the road should be far enough away for maximum focus distance on a smartphone (with mine, everything from about 6m away is "distant" and things up to 2m still look reasonable - if you get any closer at least from a car, I'd think fuzzy images are the smallest problem with that).
Not using auto-focus would make taking photos much faster (thus making the dynamic frequency of photos more useful), and avoids the frequent out-of-focus images on devices with laser auto-focus like the LG G4.
In another issue it was said that you can manually lock the focus, that seems like a possible workaround, but you'd still have to remember to do that every time and thus is really not ideal.

zstadler, 2017-03-04:
I would like to add support for this issue. 

Moreover, manual focus holds for less than 10 seconds.

kalmanbencze, 2017-06-15:
so the implementation is ready for infinity focus, together with the new camera 2.0 API which also brings some enhancements. It will be pushed on the beta channel in the upcoming days if validation goes as expected.

bogdans-telenav, 2017-12-04:
Will close this since it seems that this was delivered. If you still have issues with this please reopen.

closed 2017-12-04 by bogdans-telenav