kartaview/android #76: [question] Where is the dead reckoning code?

mvglasow, 2017-01-28:
According to the article on [OpenStreetCam](http://wiki.openstreetmap.org/wiki/OpenStreetCam) on the OSM wiki:

> The OpenStreetCam apps optionally connect to an OBD2 dongle in the car. The apps read speed and angle information to enable more accurate traces and dead reckoning in tunnels and other GPS-challenged locations.

Going through the code, I see that the app collects all kinds of location, sensor and OBD data but I can’t find where these are actually being put together to extrapolate from the last GPS location. Any hints?

gdt, 2017-01-31:
Also, it would be really cool to hoist this somehow into a location provider that merges GPS and other sensors (OBD2, builtin phone acceleromters if cal passes), for broader use.   Perhaps it could be passively listening to GPS so that listening to GPS causes it to work.

mvglasow, 2019-04-22:
I have seen some traces in tunnels which seem to follow however the tunnel was mapped in OSM at the time. Suspicion: dead reckoning simply works by taking the last known position, distance traveled since then, and tracing the road on the map for that distance.

This might work well if the map is accurate, e.g. when the tunnel was mapped in OSM based on official (e.g. cadastral) data. However, if mapping is approximate, so it the position. Most importantly, locations obtained by dead reckoning are not suitable for mapping (as that would mean feeding the map’s own errors back into it).