kartaview/android #82: Feature change request:  eliminate the popup points screen

jfburke3, 2017-03-08:
Please consider doing away with the "points" screen that pops up when you stop a sequence.  It is very distracting when driving, and doesn't add much value to shooting images.  It's not like anyone is going to be keeping a running total of points while driving.

kalmanbencze, 2017-03-23:
Hmm, this could be solved with a checkbox like "Don't show again". I'll see when this can be added to the app.

jfburke3, 2017-04-03:
It looks like a setting switch was added for this, and it works perfectly fine for me.

kalmanbencze, 2017-04-03:
yes but the switch will disable the points system also, together with the leaderboard and estimated points in recording

jfburke3, 2017-04-03:
Oh.  Well, I don't want *that*!  Thanks for pointing it out.

kalmanbencze, 2017-04-04:
It will still be calculated on the server, so you don't loose the points, you loose the functionality on the phone only, which means points and leaderboard will not be present in the app.

jfburke3, 2017-04-04:
Ah, thank you for that clarification.  While I would _like_ to have that information available in the app periodically, it is less important to me than not having the "points" popup after I stop a sequence.  (And yes, I can see that other people might like to see that, but for me, it's just another thing I have to touch the phone for before I can start a new sequence.)

kalmanbencze, 2017-04-21:
Until we add the option to not show it next time, you can disable the switch, and enable it when you're interested in the leaderboard stuff.

jfburke3, 2017-05-16:
My vanity forces me to admit that I *am* sometimes interested in the leaderboard stuff, but usually only when uploading.  When I'm taking pictures, that's less important to me, but only because avoiding crashing is more important.

I can live with the enable/disable points setting.  (Pun intended.)

bogdans-telenav, 2017-12-04:
If you have further issues regarding this please file a new bug. I will close this since it seems that your problem was already fixed.

closed 2017-12-04 by bogdans-telenav