kartaview/android #83: Increasing the adoption of OSC.

chrisdebian, 2017-03-12:
Hi, all.

I'd love to see more people contributing to OSC, and increasing the breadth of the coverage. Given that there is a huge number of people contributing to OSM, does anyone have any ideas for getting more of these people to record their journey's?

Maybe we could list ideas here, and then the Devs can comment/ discuss.

Thanks,

Chris.

davidak, 2017-03-23:
People need to know OSC, so it should be promoted on OSM.

People should see the benefit of OSC for OSM. So make changes to OSM visible that where only possible with OSC.

The best promotion might be just to use it yourself and talk about how it is useful for everyone.

kalmanbencze, 2017-04-21:
We're still trying to figure the "promote" part out. But this thread would be a good idea. Unfortunately as a dev I have very little insight into proper promotion, but my colleagues have a few ideas and initiatives that are slowly improving on this front.

stephan75, 2017-04-23:
IMHO we need a communication channel for end users who are not familiar with github etc.

Have a look at that sections on the websites by graphhopper.com or mapillary ... They have an own user forum software.