kartaview/android #84: Fixed bearing

james2432, 2017-03-13:
Phones might not have compasses, so having an option to set a fixed bearing and could either be calculated during the trip(exif edited) or on upload(would have to order all images by capture time).

Essentially it would be:
(Bearing + Fixed angle) Modulus 360

       315    0     45
             \   |    /
      270-        -90
             /   |    \
       225  180   135

Calculating bearing is pretty easy given two gps coordinates in degrees:
https://github.com/osmottawa/OSVUploadr/blob/0.1.3-ALPHA/OSVUploadr/src/main/java/ca/osmcanada/osvuploadr/Utils/Helper.java#L286

kalmanbencze, 2017-04-21:
In case we can't collect orientation data, we should calculate bearing you're right. I'm not sure if it's implemented already but bearing should be a fallback strategy on the backend. Added to backlog.

james2432, 2017-04-21:
Not only calculating bearing, but being able to set it: (say you are pointing your camera @ 30 degrees to capture more POIs or even 90 degrees.  Being able to set it for the pictures will help users decide which "trail" they want from JOSM (side view? front view? back view?)

kalmanbencze, 2017-04-21:
maybe selecting the orientation at the beginning of the track like: front/left/right/back, but having a dial for setting the orientation seems overkill, also besides the 4 orientation types, values such as 30 degree to the right are discouraged.

james2432, 2017-04-21:
so that would mean all cameras would have to shoot in 180degree coverage to enable not missing details such as store signs and other road side details... the mid angles can be useful for capturing details at lower resolution (compared to 50 feet away where you only see a pixelated mess)