kartaview/android #86: App closes frequently

jfburke3, 2017-04-03:
I'm not sure how else to describe this except to say that the app exits to the desktop (phonetop?) frequently.  There is no error message or crash message.  It just simply exits.  I'll be shooting an image sequence while driving and everything is going fine, then I'll glance at the phone screen and see that it's on the desktop.  I can relaunch the app and start again, but the exit-to-desktop is very frequent and very very annoying.

On a few occasions, I managed to get a look at the phone just before it exited, and I noticed that the camera view was frozen on a particular shot, and it stayed there for several seconds (5, 10, 15? not sure) before exiting.  I can't say that this is true for every instance, though.

OSC version 2.0.4, Android 5.0, Samsung Galaxy S5 Active.

Over the next few days (depending on bandwidth) I'm going to upload some image sets that I took today.  If you look at them, you'll notice large gaps.  Those are the places where it exited.

jfburke3, 2017-04-17:
I did some experimentation with settings in order to try to figure out what might be causing the problem (and it was a big problem--the app was essentially unusable due to how often it would crash).  I tried various combinations, and eventually found that by disabling "show map while recording" and enabling "JPEG recording mode" it almost (but not quite) eliminated the crashes.

I'll note that I haven't tried the combination "show map while recording" disabled and "JPEG recording mode" disabled yet, but that's going to be my next attempt.  (In other words, return to the default settings except with "show map while recording" disabled).

kalmanbencze, 2017-04-21:
The maps sdk had some serious issues, and still has some nasty native memory usage which affects the recording negatively. Hopefully with the 2.0.8 we've managed to fix most of the big issues.
also resolved some thread handle overflows during drive, and there will be further improvements done on points calculation and position matching.
I suggest to try and test the new beta. It should be more stable and also has some shiny new error logging features which enable us to fix 2 times more efficiently the problems our users are experiencing.

In the last 3-4 week period we were only improving the stability of the app and will continue to do so for the upcoming 2 weeks as well.

Hope to hear further feedback, and thanks for reporting back.

fbeels, 2017-04-21:
I have the same problem,on recordings > 10 minutes it crashes.

maybe add functionality to automatically upload any existing crash logs when uploading tracks

kalmanbencze, 2017-04-24:
The functionality to upload crashlogs together with relevant data is already there. If it happens with the new builds we'll know about it.

kalmanbencze, 2017-04-24:
There seems to be still a "lot" of crashes from the map sdk, even on the 2.0.8 version.
If you experience any more problems I suggest turning the sdk off from the settings during recording and upload. This way it won't have a chance to crash. You're still going to have full functionality, only difference is the map won't be displayed in the app.

jfburke3, 2017-05-16:
@kalmanbencze If the crashlog upload function is something that needs user intervention, then I have to say that I've never seen a notification about it.

kalmanbencze, 2017-05-16:
@jfburke3 no, the crash logging is automatic, but there is also the manual reporting button, for this functionality a crash or error is not needed, you simply press the send issue report from the settings, to send us the logs and a description about the problem so that we can adress the malfunction.

jfburke3, 2017-05-31:
With the map turned off and using jpg recording mode, I don't seem to have any crashes at all.

bogdans-telenav, 2017-12-04:
I will close this.

closed 2017-12-04 by bogdans-telenav