kartaview/android #88: Leaderboard: more divisions

fbeels, 2017-04-14:
When looking at the leaderboard as a new user in canada i see 2 options list the top users in the world or top users in the united states. Obviously Canada is missing,but beyond that having divisions for each province would be more motivational for people in smaller provinces (example there are more people living in downtown toronto than there are in the entire easter provinces populations combined) otherwise they would be buried in a list of all canadian users and unlikely to continue contributing after a short while if they are in 506,834th place in all of canada vs say 12th place in their own province.if the carrot (being in the top 10) is just out of reach people will contribute more than if the carrot is sitting on the moon with no chance of getting anywhere near it.

I would imagine it would be similarly beneficial for state by state divisions for american users.

chrisdebian, 2017-04-15:
Hi, fbeels.

I live in the UK. When the Leaderboard starts, the second tab briefly shows the US, but then (once location checked?) lists UK users. I would support your request for listing  'by country' initially, and as a second priority, by smaller area, such as state/ province/ county.

Can the devs say what options are currently available for the 'local' tab, is if the US/ UK? What are the future plans?

Cheers,

Chris.

kalmanbencze, 2017-04-21:
So what happens now is, we assign a 'region' to users based number of photos in regions (we choose the region with the highest number of photos as the user's region) then together with the global leadeboard this value is retrieved from the server. then we display a leaderboard for that region when the user taps "US" or "Germany" etc.

Adding more granular regions would be tricky, but we definitely want to do it, although not in the short term.

closed 2017-04-21 by kalmanbencze