kartaview/android #92: App restarts after selecting map overlay

StaufferJm, 2017-04-26:
Hey team. The map restarts after I select the map overlay so I can see what routes have been mapped. After it restarts the pop up with the signed driver comes up and restarts the app again without the map overlay.

StaufferJm, 2017-04-28:
So, after trying many different things (removing different setting, restarting, reinstalling) it won't let me log back in my Facebook account. I created a new log in using my gmail and the map overlay works! I was hoping I can transfer my data from when I was logged into FB to my gmail account. Is this possible?

StaufferJm, 2017-05-09:
Has anyone looked at my issue? Two weeks and no responce?

jfburke3, 2017-05-16:
Not one of the developers, so please don't get mad at me.  I have to say that I'm not quite sure what you mean by the initial description.  Can you be a little more detailed?

kalmanbencze, 2017-05-16:
So what I assume is happening is that @StaufferJm is a contracted driver, maybe you can confirm it @StaufferJm . Then the account that he created through our driver program is a prof. driver account which disables map and points, etc so the app can record and do it with no crashes. What the "restarts after I select the map overlay" means, I dont fully understand. Maybe you can provide a little clarification.
Also I'm sorry about not replying instantly but we are 1 developer on each project only so you can imagine we are quite busy. I try to respond to issues as often as I can.

StaufferJm, 2017-05-16:
Thanks for getting back to me! So if I'm a contracted driver map overlay and points don't work? If that's the case then there is no issue. I didn't know that. I have been able to log back into my contracted account via Facebook so there is no longer an issue. Thanks again for responding.

closed 2017-05-16 by kalmanbencze