kartaview/android #94: Sensor data basic question

vr00n, 2017-05-13:
We are a student project trying to integrate OSC images with the underlying sensor data. Currently we use track.txt to do this. But we found that track.txt gets deleted after uploading to OSC servers. 

Can you point us to some best practices please?

Thanks!

kalmanbencze, 2017-05-16:
Hello @vr00n , I have checked out your 3 projects, they seem interesting. the tracks.txt.gz files get up[loaded then deleted, that I cannot modify, the space needs to be freed after upload.

Can you be a bit more specific about what information do you need?

Best,
Kalman

vr00n, 2017-05-18:
Thanks @kalmanbencze . We would like to access/preserve. at least location (lat,long) and accelerometer data (x,y,z). Are these available via the API - if so would appreciate how to access. Thanks!

kalmanbencze, 2017-05-18:
after upload, you can access the photos and metadata file link through the api, i think the method is called details, but you'll have to parse the data from the file, using the format I've attached the other day

kalmanbencze, 2017-05-18:
lat lon can be accessed with every image, but the other sensors aren't copied to the DB, so no way accessing it without parsing

kalmanbencze, 2017-05-18:
if I'm correct, but should ask around on the openstreetview.org project

closed 2017-05-18 by kalmanbencze

vr00n, 2017-05-18:
Thanks @kalmanbencze . This is being discussed in https://github.com/openstreetcam/openstreetview.org/issues/109