kartaview/android #95: Easy ways to optimize energy usage?

gappleto97, 2017-05-28:
Hi,

So I tried your app today. I'm currently uploading ~16km worth of pictures, so yay.

Unfortunately, this app caused my phone to discharge *while plugged in on USB C* .

Now, this could be a one-off problem, and it could be nothing. But I'd like to use this on road trips and the like, so it would be nice to investigate how this caused my phone to draw more than 15 watts.

kalmanbencze, 2017-05-28:
Since the app uses a lot of sensors + camera and video encoding, it uses a lot of energy. 
I can suggest 3 easy ways to optimize the usage, some are with compromises but here it goes:
-Shut off the screen of the phone during recording, it will continue to record in the background, no compromises here but visuals. 
-consider switching to JPEG files instead of mp4, this uses less CPU thus conserving more battery power. The drawback is that the track's size will be larger, so upload will take longer. (there is a setting on the settings screen for it) 
-disable Gamification during recording. This also conserves some CPU "heavy" calculations. This will only affect your ability to preview your points locally, after upload you receive the points regardless of how you recorded the track.

gappleto97, 2017-05-28:
> -Shut off the screen of the phone during recording, it will continue to record in the background, no compromises here but visuals.
-consider switching to JPEG files instead of mp4, this uses less CPU thus conserving more battery power. The drawback is that the track's size will be larger, so upload will take longer. (there is a setting on the settings screen for it)

I did both of these already. Was discharging on power despite these.

> -disable Gamification during recording. This also conserves some CPU "heavy" calculations. This will only affect your ability to preview your points locally, after upload you receive the points regardless of how you recorded the track.

I'll give that a shot, thanks. Nonetheless, I feel like there's some optimizations here that really need to happen.

gdt, 2017-05-28:
Or perhaps, dynamic adjustment of picture interval, in order to ensure that the phone is either >95% or having positive charge.

gappleto97, 2017-05-29:
@kalmanbencze, that last one put it just over the threshold I needed to find. Before disabling that I was discharging (while plugged in) at ~1% every 5 minutes. After disabling gamification it charged at ~2% every 5 minutes. That's enough that I should be able to have a satnav open for a road trip. It would be nice if I could have that without disabling so many settings though, so again, I'd say that optimizations would be useful here.

kalmanbencze, 2017-05-29:
Can you please send us a bug report with the app version and details
written in it (from settings)? What device, android version etc? Also what
build were you using when it was discharging quickly. If you were using
mobile data, if you saw the coverage data during drive?
Thanks, appreciate it.
Kalman

jfburke3, 2017-05-31:
I have the same problem.  I can plug it in to AC power and the battery still runs down while uploading.  In addition, my phone gets pretty hot.  I frequently will pause or cancel an upload due to the heat, in order to let my phone cool down.  Turning the screen off doesn't help.

Note that this is during *uploading*, not while capturing images.  

Android 6.0.1 running on Samsung Galaxy S7 Edge.

kalmanbencze, 2017-06-06:
I can give you a simple explanation to why this is happening but I might be wrong, or it might not be satisfactory to you.

So all in all, the problem is misuse of the phone hardware. Basically these phones aren't made for data collection and continuous picture taking over long periods, and also they are not streaming machines. 
Basically their hardware is for lightweight web surfing and mainstream usage.

So using phones to gather image and sensor data is not optimal but is much cheaper than designing and manufacturing hardware then distributing it with a pricetag. It's not perfect but it -sort of- works. Iphone on the other hand has a lean on software-hardware, it's much more optimised and both their hardware and software is more premium than anything on android.

Specs don't matter, real life performance, efficiency, and optimally written kernel are the advantage of the iphone. But it also costs $800.  

Non the less I'm looking into these issues, there is still space for optimizations and improvements. clearly it can be done better, it's just not goint to make a huge difference.

gdt, 2017-06-06:
Another way to look at this is to view draining battery while on power as unacceptable, and have throttling in terms of picture taking and/or processing to keep the system at least neutral.  Really, I'd want to charge to at least 90% and stay their; my personal usage pattern expects to get charged during a significant drive.    It might be that taking a still image once a second (or once every 10m, if that's less) would use less power, and it might not be all that less useful.

gappleto97, 2017-06-06:
All I can tell you is that Mapillary doesn't have this problem, even when I change its settings to be more intensive. With this app it's like that right out of the gate.

kalmanbencze, 2017-06-06:
mapillary uses jpeg if I'm correct, also they don't log pressure, gyroscope, accelerometer, gravity, compass.
I'm currently looking for ways to improve the performance and optimize the battery usage. I'm looking into alternatives for FFMPEG for encoding the high res videos. FFMPEG isn't exactly optimized for mobile, we were using it because of backwards compatibility.

jfburke3, 2017-06-06:
Mapillary does use jpeg, but I have JPEG mode enabled in OSC, too.

kalmanbencze, 2017-06-06:
so until now:
- optimize local points calculation
- with camera2 api comes better encoding - not using ffmpeg
- check sensor data collection how much impact it has on battery life

gappleto97, 2017-06-06:
Is there any reason you can't move some of that processing to the upload time? That would almost certainly lower battery impact, especially since I imagine most people upload while plugged in.

jfburke3, 2017-09-26:
@gappleto97 The problem is that my phone battery drains (and gets pretty darn hot) while uploading, too.  Not sure that would be a good solution, unless uploading is throttled.  As it is, I have to take breaks during uploading to let the battery recharge (it frequently drains down to 0) and cool down.

naoliv, 2018-08-29:
What exactly OSC does when uploading the files?
The phone gets extremely sluggish, hot and drains all the battery.

alexandrui-telenav, 2018-08-30:
We're planning to refactor a big part of the upload code to improve quality and performance.
We're not mining bitcoin in the background, only thing that happens is uploading files with full speed.