kartaview/android #97: Frequent "upload canceled" when trying to upload

jfburke3, 2017-05-31:
With OSC version 2.0.11, I've begun encountering frequent "upload canceled" issues.  I'll go to the Upload screen and select to upload all of my tracks, it will change to the upload progress screen for a few seconds, then the message "upload canceled" appears at the bottom and it returns to the main screen.  I will go through this anywhere from 3 to 15 times before it finally begins to upload images.

Once it begins to actually upload, it seems to be fine.

This is pretty consistent with version 2.0.11.  It was rare with previous versions, and even when it did happen, I only had to retry once or twice.

Android 6.0.1 running on Samsung Galaxy S7 Edge.

kalmanbencze, 2017-06-06:
Could you please reproduce the issue, then from the setting send an error report? it will attach all logs from the prevoious 48 hours. This way I can check out the response from the server, it might be a server issue if it times out or anything similar.

jfburke3, 2017-06-06:
Sent.  For the description I put:  "Upload cancelled" github # 97

I don't think it's a timeout from the server.  It happens almost immediately--too quickly for that to be the case, I would think.

jfburke3, 2017-06-06:
What you're looking for should be among the most recent events in the logs.

kalmanbencze, 2017-06-06:
Ok, thanks for reporting it.

kalmanbencze, 2017-06-08:
got the logs, found the issue. I just validated the fix. I will be pushing a new update to the beta soon which will include this.

jfburke3, 2017-06-09:
I installed 2.0.12, which I guess is the new beta version, and it seems to be uploading perfectly now.  As an added bonus, you seem to have fixed the "xx/yy" count in the upper right corner, which also wasn't working (it only showed "00/yy").

kalmanbencze, 2017-06-09:
Glad to hear it worked! The label issue was fixed a little whole ago, I
think... Might be that I didn't include it somehow in the previous build...

closed 2017-06-09 by kalmanbencze