kartaview/android #98: Very severe upload failure

gappleto97, 2017-06-01:
Hi,

The most recent track on [my profile](https://www.openstreetcam.org/user/gappleto97) uploaded *very* wrong. Somehow it got split into three separate tracks, two of which are identical, and both of those jump from one location to another that's quite distant.

The third one, which was the most accurate (though still far off), has another jump in location, and misses several roads that I went on.

All in all, it should say that I went 30 km with ~1600 images. Instead it says that I went 20 km with ~1200 images combined.

On top of that, when the upload started, it gave a percentage with so many digits it overflowed from my phone screen.

Now, the only thing I did differently from previous uses was enable the mp4 encoding scheme. I find it hard to believe that would be the cause, so I'd appreciate it if you helped me diagnose this. If you tell me what files to give you, I can see what's on my phone. But unfortunately I can't provide you with the original trace, since your app automatically deleted it.

gappleto97, 2017-06-01:
Is it possible that this is only an error while it's processing, and that it will resolve itself eventually?

gappleto97, 2017-06-01:
I would also point out that two of the tracks are marked uploading, but my phone said it was finished uploading multiple hours ago

vr00n, 2017-06-01:
Myself and @alephcero are experiencing similar issues.

OSC usernames `varun-argo` and `felipe-diego-gonzalez`

gappleto97, 2017-06-02:
For the sake of clarity: I'm going to upload more tracks now. I'm going to first back up the track directory, to make sure I can reupload if there is a failure, but I'm going to put up tracks now.

kalmanbencze, 2017-06-08:
Hello @gappleto97 @vr00n , please when you encounter this problem again send an issue report.
that way we receive the latest logs. I'll have to check them to see what happens.

@gappleto97 i checked your track, it has ~190 images on the website but it should have around ~1680 images based on the metadata file. The server didn't split your video files correctly I think. I'll check it out to see what happened.

kalmanbencze, 2017-06-08:
@gappleto97 most probably it can be recovered by re-processing the files.

gappleto97, 2017-06-08:
I won't be encountering the issue again, because I no longer intend to use your service. Its just too buggy in its current state for me to use it. I've had issues with almost every upload, and the whole experience is just too frustrating to be worthwhile.

kalmanbencze, 2017-06-08:
Sorry to hear that. It's understandable though.

vr00n, 2017-06-16:
For what its worth - we have largely had a positive experience using the app. We are fans and supportive of your efforts.

kalmanbencze, 2017-06-16:
Thanks, I we really appreciate it. I didn't forget about the issue, it's just that I don't really have much to go on. Got to reproduce it somehow, meanwhile I'm trying to improve what can be improved directly.

kalmanbencze, 2017-06-28:
@vr00n Hello, sorry for the delay. Seems that I've managed to implement a fix for this issue.
It was a weird one. Could you give feedback after the 2.0.13 release? Was it happening often for you?
Thanks

kalmanbencze, 2017-06-30:
@vr00n  The update has been posted, you can find the apk here on github under releases and install the apk, or wait untill it gets rolled out on the beta channel.

bogdans-telenav, 2017-12-04:
I will close this. If the issue is still reproducing please reopen.

closed 2017-12-04 by bogdans-telenav

vr00n, 2017-12-04:
Apologies @kalmanbencze - I missed this notification. Since you updated, we have seen pretty stable uploads although we were on more robust wifi. 

Thank you.