kartaview/android #99: Link a 360 Camera

DonZaz, 2017-06-03:
Hi Everyone, 

There are so possibility to use the app through an external 360 camera? Like the app of Google Street View?

kalmanbencze, 2017-06-06:
Hi @DonZaz ,
There was an experimental implementation working with 2 external 360 cam in the android app, but it was early on, and wasn't exactly the fastest thing because of the image transmission over wifi, so it didn't get into the app. Recently there was added support on the front end to check out 360 pictures, but it was taken out of the app.

So right now there is no way to record with external 360 cameras. We might add support for theta, theta s and all the newer open spherical camera API compatible cameras but not sure when.

DonZaz, 2017-06-06:
Dear OpenStreetCam,

Would be very useful reinstate the option to add at the least one camera 360.

I am doing this with Google Street View now, that was quickr, as i am not yet a developer.

I can spend my time doing this and helping you guys, I am a professional surveyor and for me is the right cause

I though in some way we can improve our maps, also with constantly update through these little devices

Even more with ProfessionaL Devices, that not only willa allow us to have an high quality Street Views but also we can have a realistic dimensional data with other products, as the Leica mini Laser Scanner. BLK 360.

if there are some way for helping further the community, I would be glad to get help learning API

Enrico Lazzara

Architect and Project Manager– London

City of London SW2 3BX

Work +44 7446222756

geom.lazzara@outlook.com

geom.lazzara@pec-mail.eu

Linkedin.com/in/geoengtechel

Follow update on twitter

PARIS - LONDON - MONACO- USA- CANADA - MIDDLE EAST - CHINA - HONG KONG