kartaview/android #100: No SDCard Storage Option Showing on Samsung Galaxy Ace Style

TallGuySK, 2017-06-15:
Device: Samsung Galaxy Ace Style (SM-S765C)
Android OS: 4.3
OSC Version: 2.0.12

I am attempting to use OSC with the Samsung Galaxy Ace Style. I have been able to get OSC to record directly to the device, but the option to turn on SDCard Storage does not show up in Setting Menu.

SDCard has been reformatted, tested in OSC other devices, and compatible with other Apps on this device.

Any thoughts? Is is a Device or Android OS Issue? Any way to enable this?

kalmanbencze, 2017-06-16:
Hello @TallGuySK
For pre Kitkat devices the method of detecting external storage space is different. Tried to do a blind fix (unfortunately I dont have the device and I couldn't reproduce it on other devices), to try to detect using 2 different methods, I will push an update on the market in the upcoming days, hopefully you're device will show the external storage option after that. Please post the results here after you get the 2.0.13 version.

TallGuySK, 2017-06-28:
Any update on 2.0.13?

kalmanbencze, 2017-06-28:
Sorry it got delayed a bit. We're doing the final steps towards finalizing it. I will let you know when it's available on the beta channel.

kalmanbencze, 2017-06-30:
@TallGuySK The update has been posted, you can find the apk here on github under releases and install the apk, or wait untill it gets rolled out on the beta channel.

TallGuySK, 2017-06-30:
@kalmanbencze Installed 2.0.13 APK. Still no SDCard Storage Option on this device.