kartaview/android #102: canot login with OSM

grinapo, 2017-08-09:
This has been going on for a while now, and present in the beta as well. (And you just got an android creash report, too.)
I cannot login by OSM oauth. Selecting the OSM button opens an empty white rectangle (I suppose it's kind of web display), which usually hangs there. Sometimes display OSM login page then freeze after the name, or password, or maybe by selecting the checkbox, then the application usually crash, if I wait for a few minutes.  Now I was even able to push the login button (unexpected feat), making it crash 40 seconds after that.
Obviosuly it doesn't quite help uploading the imagery waiting.

grinapo, 2017-08-09:
But that's funny. Last crash actually logged me into OSM, so after 2 further crashes I was able to select "grant" on OSM login page (since I've been logged in 2 crashes before), and in the next crash I am logged in for now. I've been trying for a month or so. 😉

grinapo, 2017-08-09:
…so now the upload is finished. In the top 1/10th horizontal band of the screen there's the map. Below that 2/10th is a dark gray stripe of uknown purpose. Below that there are the list of the uploaded imagery as if it wasn't uploaded (but the android status line said they were). Since there's no menu or else button I can only leave the app by the back arrow.

Starting the app again goes full dark gray for 30 seconds, then seems normal. I'm still logged in, yay.

kalmanbencze, 2017-08-09:
Hey, thanks for reporting, will check these out tomorrow, a fix is long
overdue for this if you're experiencing it for at least a month. Just
ignore the broken ui after the background upload, some states have been
lost, that is why it's weird.
The login issue seems weirder, could you send us a bug report, mentioning
any relevant info? Are you using a custom rom? Maybe the beta chromium
webview? Anyways, I will check it out and post an update tomorrow. Cheers

kalmanbencze, 2017-08-10:
Could you please report an issue throught the settings? Under Improve/Report a problem. Thanks!

grinapo, 2017-08-14:
@kalmanbencze sent you a report right now.

kalmanbencze, 2017-08-16:
Forgot to mention to reproduce the issue before the issue report, because we send the logs from the last 48h only which means you'll have to log out then try to log in again, then send the report. Also try to wait a bit longer when doing the login. The webview is missing the loading indicators between the 2 phase login ui from OSM so it might be just taking a bit longer. Depends also on the connection.

Eventually try again with the new nightly build when it gets released.

grinapo, 2017-08-17:
Okay. So now you got two reports: one from android (since osc have crashed after 30 sec on the login screen) and the other from inside the app.
The net connection is fine here, the page usually comes up in the browser within 666ms.
And I will, eventually, try again since now I'm logged out forever, again. :-)

kalmanbencze, 2017-08-17:
Okay got the logs, found the reported crash. The stack trace doesn't tell me much, only that the webview implementation is crashing. Try using another webview implementation (you can change it settings --> developer settings --> webview implementation) if there is an alternative available, like "Chrome stable" select it. Unfortunately the webview is not part of our app, it is part of the rom. You might try logging in again and just staying logged in and wait for the next nightly or beta from lineage os, or use Oxygen os.

closed 2017-08-17 by kalmanbencze

grinapo, 2017-08-18:
Nope. I have sent you the crashes with chrome stable webview just now.
(Technically you're in the position to report this to upstream since you have the data.)

kalmanbencze, 2017-08-18:
Report this where? I dont understand.

kalmanbencze, 2017-08-18:
Could you try disabling the sdk from the settings, restarting completely the app from the notification at the bottom of the screen, then try to log in?

grinapo, 2017-08-21:
Report from a) the android os feedback of crashed app, and b) from withing the app "report problem"

Re disabling: where exactly want do you want me to disable which 'sdk'? I can kill app and services fine, this happens to be a well rooted phone. :-)

kalmanbencze, 2017-08-21:
What I don't understand is "upstream". I got your reports but it's just a plain webview crash, can't really fix that.
The disable sdk switch is in our app's settings screen. After disabling the switch and restarting, the app will work without displaying the map. Sometimes the map sdk affects performance and stability, that's why the switch exists.

kalmanbencze, 2017-08-21:
*Also the stacktraces from the report are incomplete unfortunately

grinapo, 2017-08-23:
"Upstream": you say webview is crashing; either it's a problem with OSC crashing the webview (so you have to fix) or it's a webview bug (then you have to report it to whomever maintains the webview, since only you know what and how you wanted from webview). In light of OSD crashin both webview implementation I'd say it's probably not webview's fault. But I seem to have reached a point where I spend more time to convince you to do something than providing any useful info, so I better leave now.

"disable sdk": if it's there then maybe we have different apps.  I have:
* account: login
* general: upload, auto upload, distance, enable map, show map, points
* cam: resolution, jpg, fix focus
* obdii: whatevers
* improve: report, send feed, tips, walkthrough
* about: 2.1.1

There is no SDK. -- Neo

kalmanbencze, 2017-08-23:
Sorry the option is called Enable map, its under General.

The webview is natively crashing which is run in the applications process, thus crashing the whole application, so there's that.
Since you are using a nightly of a custom rom I don't think this is has priority over other features, it is going to be fixed in the stable version. When you started using nightly builds you implicitly agreed to put up with any bugs. It is unstable. If you want stability use their latest stable release. I can't report the issue because chromium has no issue here, its the custom ROM's fault, and if you'd like, you can head over to xda or send a message to the lineageOS devs. Event if I sent them a report they would be unable to implement a fix because there is no stack trace available from google developer platform.

To conclude, this is not an OSC app bug.

grinapo, 2017-08-24:
Thanks for your time.