kartaview/android #106: Upload with mobile data instead of wifi

mentalo-, 2017-08-17:
Hi there, 

When i upload a track, this is very slow because my wifi (DSL) connection is bad. I reach 80kbps max, one track takes hours to be uploaded. My mobile connection, in 4G, is faster than DSL. But when I disabled wifi, I cannot upload in 4G, there is a message telling me to activate wifi.

Is it possible to choose the connection mode, depending of the speed of each ? It would be useful.

kalmanbencze, 2017-08-17:
Hello mentalo, there is a setting called "upload on data" in the app's Settings page, if you enable that, you can upload the collected data using your cellular network connection. 
Cheers!

closed 2017-08-17 by kalmanbencze

mentalo-, 2017-08-17:
Ok ! Thanks kalmanbencze, I use the app for a few days and don't know it very much yet.
Sorry for the fake issue :)

kalmanbencze, 2017-08-17:
No problem, happy to help! :)