kartaview/android #107: Allow uploading to mapillary too

moy, 2017-09-03:
Hi,

That may sound like a strange request, but ...

Like probably many other users, I'm happy to contribute images but 1) I'd like to ensure survivability of my images as much as possible (i.e. make sure my images don't just disappear if ever OpenStreetCam dies), 2) I can't really choose between OpenStreetCam and Mapillary. So, the best option to me would be to upload to both services at the same time. An alternative would be to copy images already uploaded to Mapillary to OpenStreetCam [as suggested here](https://github.com/openstreetcam/upload-scripts/issues/13). Another one is to do the upload manually using the upload script. But a simple option to do everything from the app would be really nice.

I believe this could help OpenStreetCam grow: currently, if I had to choose only one service, I'd probably choose Mapillary which is currently more advanced (more complete image set, good integration with openstreetmap, ...), but I'd really prefer contributing to both at the same time. I think I'm not the only one to do this reasoning.

Thanks,

Skippern, 2017-10-07:
I currently process all my sequences done by Action cams with scripts and send to both. The rare occacions I use the mobile app I end up only contributing to Mapillary, an `upload to multiple services` option would make it much easier to contribute to multiple platforms, and thus ensuring a larger userbase to all of these projects.

This is just as valid point for the `iOS` app.

tuxayo, 2018-06-04:
https://github.com/openstreetcam/upload-scripts/issues/13#issuecomment-368335088
@adb01730
> Is there a way to upload my openstreetcam pics to mapillary?
I was using the mapillary app until if failed to detect the gps signal after a android update. I switched to OSC which I now prefer to mapillary. The OSC uploader provides % complete, time to complete, and KB/s. With Mapillary I had to monitor the upload because it would often crash after several hours and have to be restarted.

tuxayo, 2018-06-04:
https://github.com/openstreetcam/upload-scripts/issues/13#issuecomment-368518319
@alexandrui-telenav said:
> @adb01730 Good to hear your feedback!
It is possible to copy recorded data from the OSC app before uploading, but the optimised format we're using is not compatible with Mapillary as far as we're aware.

> In order to optimise upload time, we encode the photos as frames in video files ( this reduces upload size 2-3X compared to individual JPEGs ). One could split the video files into frames and geotag them based on the metadata file before uploading to Mapillary, but it doesn't work out of the box.