kartaview/android #108: Please clarify the licence of this repo

moy, 2017-09-07:
Hi,

The OpenStreetCam is said to be open-source in several places, but I can't find a reliable reference for this. Can you add a LICENCE file to this repo to make it clearer (like what's done for the iOS app for example: https://github.com/openstreetcam/ios/blob/master/LICENSE).

Thanks,

alexandrui-telenav, 2017-09-11:
Hi @moy, you're right, this should be clearer. Will add the same licence file here.

moy, 2017-09-12:
Thanks. I'm writing an article on OpenStreetCam that I'd like to publish ASAP, so it would be cool if you could add the file before I publish it, so I can link directly to the license file.

kalmanbencze, 2017-09-12:
Hi @moy,
Just added the license file, you can go ahead and publish.
Thanks, 
Kalman

closed 2017-09-12 by kalmanbencze

chrisdebian, 2017-09-12:
Do you have a link for the article, it would be good to read?

Cheers,

Chris
chris_debian

gdt, 2017-09-12:
@moy: You may want to check also about the licenses of the binary apps.  As I understand from earlier discussion about this being on f-droid (https://github.com/openstreetcam/android/issues/8), there are proprietary libraries and thus the apk is not Free Software.   And of course, nothing obtained from Apple's iOS store is Free Software; at best it was Free on the way in -- but that's not about openstreetcam.

moy, 2017-09-12:
The article is here: https://linuxfr.org/redaction/news/cartographie-et-images-de-rues-mapillary-et-openstreetcam

But 1) French-only, 2) you need a Linuxfr account to see the draft (it's free), and 3) the URL will change when I submit the news (it will be public afterwards).

chrisdebian, 2017-09-12:
Thank you.

Chris.

moy, 2017-09-14:
FYI, final and public URL of the article (still french-only): 
http://linuxfr.org/news/cartographie-et-images-de-rues-mapillary-et-openstreetcam