kartaview/android #112: In-app upload hangs at 3.9MB

jfburke3, 2017-12-06:
Phone: Samsung Galaxy S5
Android ver: Android 6.0.1
OSC version: 2.1.5

Problem:  the in-app Upload function is stuck.  I tell it to Upload All, and it gets to 3.9MB uploaded before it just seems to hang.  The throughput rate is excellent (>1MB/s), and remains at a good number for hours, but no further upload takes place.

This is the first time I've encountered this.

alexandrui-telenav, 2018-06-25:
Had some problems during the weekend with the database server. Is it still hanging? If yes can you please Report a problem from Settings? ( sends logs to debug ).

deadice, 2018-06-25:
Hi, I tried and it froze there too lol.

 I have still been trying to upload tracks all day and it won't finish.

ciprianh-telenav, 2018-08-08:
@jfburke3, @deadice, do you still face this issue on latest version (2.3)?

deadice, 2018-08-08:
Yes, but it isn't as bad now. I had a long trip (500km) the other week
going down back roads and it took 2 days to upload the 12gb of data. I had
to frequently restart the upload or restart the phone to resume the upload.
Other times upload will be stuck at 5kbps.

Generally I can upload my tracks overnight, but not all the time. Last
night only 50% uploaded out of 2gb.

The issue is not on my network as I have a solid reliable connection.

Is there a beta version of the app I can use to try?